prusa3d/Prusa-Firmware #1967: First layer calibration under extrusion issue

rdpeake, 2019-07-06:
I have left a comment in the forums as well - but thought i should report here as well.

I had an mk3 and it was working as expected - i then got an MMU2 and mk3s upgrade and ever since the upgrade it has not been working correctly. When running the first layer calibration i get the following issue (with or without mmu2 connected) the purge process looks fine with decent filament flow, but the moment it swaps to the actual test proceedure it starts to extrude as expected and then quickly drops to under extrude - to the point it is impossible to get a decent first layer calibration as there is not enough plastic comming out to stick.
[image]
[image]

Prior to upgrading, the hot end, and filament originally used was working fine, and it was sticking to the bed. I also don't see any residue on the bond tech gears (the printer isn't an overly used printer, maybe 1.5kg of plastic has gone through it.

I have replaced the hotend completely, used brand new filament, cleaned the bed as best i can - all to attempt to get it to stick. i Also disconnected the MMU2, and used only a short piece of PTFE tube just to protect the filament from the metal insert in the hotend of the mk3s+mmu2. However the only way i was able to get a successful print to work was to overextrude by 10% in slicer settings (1.1 multiplier). Now i am sure the 10% is likely to be more than i needed for it to work, but i was just going for, can i get it to work at all. Is there an issue in firmware (either because it was a mk3, or otherwise) or do you have any other suggestions as to where to look for why the first layer wizard (and normal prints) seem to under extrude?

mws-rmain, 2021-05-10:
I have a similar issue (tho I'm a new user). I have an MMU2, but have not installed it yet. The 'First layer calibration' routine is in firmware, so we have no control over the setting used for this process - tho it's great you were able to find adjusting the extrude multiplier helps your prints. 
My 'First layer calibration' results look like @rdpeake; the 'purge bar' looks fine, the start of the 'zig-zag is missing, then extrudes well, but soon tapers down to significantly less ('under-extruded'?).
Have tried replacing the nozzle.
Calibration height seems to be around 1.190mm, but sketchy results are common during this routine.

rdpeake, 2022-05-01:
Coming back to this now, as i'm back up and running again - my resolution was to change to a smooth sheet. there may have also been some issues with tension in the filament line, but as i no longer have this issue i can't confirm if that was the cause

closed 2022-05-01 by rdpeake