prusa3d/Prusa-Firmware #1969: Stopping a print very early, doesn't let printer start a print again.

maximlevitsky, 2019-07-06:
The printer was executing the following gcode I use, it was waiting for pinda cooldown, and then I stopped the print. Later I coudn't start a print again unless I reset the printer.

; Initial settings (same as in prusa gcode)
G90  ; use absolute coordinates
M83  ; use relative distances for extrusion

; Wait for PINDA cooldown
M104 S0   ; turn off extruder heater
M140 S0   ; turn off bed heater
M106 S255 ; turn on fan
M860 S35  ; wait for PINDA cooldown <------------
M107      ; turn off fan

ManuGithubSteam, 2019-07-07:
I see a similar issue. Then the bed heats up the printer can not be stopped (via octoprint) while the bed is heating. Canceling the print is only possible when bed fully heated.

haarp, 2019-07-07:
You might want to strobe the DTR line on the serial interface when starting a print, to reset the printer before each new print.

maximlevitsky, 2019-07-22:
Actually I wasn't using octoprint. I just forgot to insert fillament or so and stopped the print from the menu,
Btw it would be nice to ask user to load fillament if nothing is loaded on print start, but I guess rom space is better used for something better.

closed 2023-09-28 by github-actions[bot]