prusa3d/Prusa-Firmware #1971: Filament runout doesn't trigger when 'sensor info window is open'

maximlevitsky, 2019-07-06:
This is very surprising when you try to debug the filament sensor. Happens always. Closing the menu triggers the runout immediately.

ManuGithubSteam, 2019-07-07:
This is not true for the mk3s. I testes this on two different machines. The debuge menu works as it should be you can see the number change from 1 to 0.Are you on the mk3 ? 

Side note: The sensor is flimsy and it needs to be posstioned more the the right side to get a reliable result. But after that it works every time.

maximlevitsky, 2019-07-07:
The sensor value does indeed change from 1 to 0.
However the print keeps on going!

leptun, 2019-07-07:
I think this is expected behavior as that menu is only for debugging. When you debug it might be useful not triggering autoload or runout as with that menu you test wether the sensor is working or not. I don’t see a reason to keep sensor info open during a print. This also applies for the other menus in “support”.

Also I don’t think I have ever debugged the filament sensor during a print as the firmware will always trigger runout if enabled unless you are in one of these menus.

maximlevitsky, 2019-07-07:
This is classical bug vs feature thing. I discovered this when I was trying to debug why the printer won't back off the filament when it detects a runout. I opened that window to see if it maybe doesn't act fast enough on the filament sensor and lets the filament go past the gears, and due to this misleading behavior, it indeed happened, although that was not the root cause of the problem.
Thus I wasted lot of time (and filament) on this.
For the root cause I had opened this bug:

https://github.com/prusa3d/Original-Prusa-i3/issues/135

closed 2021-02-23 by DRracer