prusa3d/Prusa-Firmware #1972: [Feature request] run fans on slow when the hotend is hot to avoid melting the printer

Mirarkitty, 2019-07-07:
==When the hotend is hot (over 160°C), make an option to force run the fans at low speed (say 10%).==

When preheating or in the first layers of a print, parts of the printers can reach glass transition temperatures. The parts that melts are the fan shroud and the lower part of the extruder front.

Having the firmware force start the fans even at low speed would stop this from happening. The parts are probably melting because of convection and even a very small breeze would help the issue.

This needs to happen outside g-code and triggered by warm hotend temperatures. Maybe both the trigger temperature and the percentage to run the fans should be possible to set.

See: https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/mk3s-part-melted/

maximlevitsky, 2019-07-07:
To be honest, the better solution for that would be to fix the lower part of the extruder to not melt - in R3 version that didn't happen. I had too to trim that part that melts.

Mirarkitty, 2019-07-09:
I don't see how parts will not start to melt if the hotend is 285°C if there's no cooling other than if they are built for something that can withstand 300°C. And that's outside the scope of self-printed parts. Printed titanium would probably be fine though.

I've had to replace my fan shroud a few times by now just from how melted it get from the first three layers of printing copolyester at 270°C. 

This should be an easy sw fix to protect the printer.

leptun, 2019-07-10:
Always running the fan at a slow speed in not recommended as there are cases where you want NO airflow around your print. Say for example you print PolyCarbonate. For best adhesion you want to have absolutely no fan blowing on the part. Same for abs. Like @maximlevitsky said, best would be for a R5 extruder redesign that fixes the problem through hardware, not a software hack.

Mirarkitty, 2019-07-10:
Well, I'm not entirely sure how a redesign of the fan shroud could fix the issue, but I'm all for that.

michalxfanta, 2019-07-16:
We carry out final tests of modified plastic parts of the extruder, I hope we publish them next week.

Mirarkitty, 2019-07-18:
Sounds good. Could I still get fans running on preheating and when temperature is set high from settings, ie when it's not printing, as a precaution?

GurliGebis, 2019-07-26:
@michalxfanta how is it coming along with the new designed parts? 🙂

Mirarkitty, 2019-07-27:
*melts another shroud*

leptun, 2019-07-27:
@Mirarkitty I don’t think I’ve ever had the R4 shroud melt on me even after printing PC at 270-280 with no fan at all. I’ll check when I get at that printer wether it started melting or not. I didn’t have any problems with cooling, so I expect it to be fine.

Btw did you print the shroud from petg or abs?

leptun, 2019-08-02:
@Mirarkitty I've checked the printer and there are no signs of melting at all. Not even a bit of sagging. I really don't know why yours does like this.

michalxfanta, 2019-08-05:
Redesigned parts for [download](https://github.com/prusa3d/Original-Prusa-i3/tree/MK3S/Printed-Parts/STL)
[extruder-body.stl](https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/Printed-Parts/STL/extruder-body.stl)
[extruder-cover.stl](extruder-cover.stl)

closed 2023-09-28 by github-actions[bot]