prusa3d/Prusa-Firmware #1977: FW3.7.1: Unable to resume print while paused in intro line

ochm, 2019-07-08:
FW3.7.1 
I paused print while executed intro line. After resume print didn't continue, no movement...

leptun, 2019-07-08:
Is this something that can be repeated? Aka did it happen more than once?

ochm, 2019-07-08:
It happened to me one time, i had no reason to do it again, but i can test it, after 3 days, i have looong printing progress.

leptun, 2019-07-09:
I don't print from the sd card so I didn't use the built-in pause function. Also my print queue is full for now. Can't help you with testing, but can ask you some more things.

Where did you pause the print exactly?
Where did the movements stop?
Can you post what is reported by the printer on the serial? (That is if you can recreate the scenario). Just use arduino ide serial monitor and connect over usb.
Are you using a modified start gcode script or the built-in defaults in PrusaSlicer?

ochm, 2019-07-20:
I can reproduce as many times as i want. The start gcode is little bit modified:
M115 U3.7.2 ; tell printer latest fw version
M83  ; extruder relative mode
G28 W ; home all without mesh bed level
G1 Z3
G1 X50 Y50 F5000
M140 S[first_layer_bed_temperature] ; set bed temp
M109 S160 ; Set extruder temp before bed level
M190 S[first_layer_bed_temperature] ; wait for bed temp
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
M109 S[first_layer_temperature] ; wait for extruder temp
M300 S440 P25
M221 S100 ; Set flow to 100
G92 E0.0
G1 X50.0 E10.0  F800.0 ; intro thin line
G1 X100.0 E6.0  F800.0 ; intro thiner line
G92 E0.0
M221 S{if layer_height<0.075}100{else}95{endif}

I do pause somewhere between intro thin and thiner line.
Extruder moves up, bed still heat and moves front, hotend stops heating. OK.
After resume hotend heats to working temperature, extruder and bed moves to paused position, and...
nothing more. Bed is on desired temperature, hotend is on desired temperature, but no moves, no print.
OK, time to stop print. Selected from menu, display shows Print aborted, but, the bed heating is stopped, but nothing more! Hotend is still heated, and not moved up... 
I try it two times with the same behavior, I can make so me video...

wavexx, 2020-01-27:
There was a long list of pause/resume fixes which is now available in the 3.9.0 (currently in beta). More likely PR which could have fixed this is #2274. Could you give it a shot?

gudnimg, 2023-07-03:
As this issue is likely stale and no longer an issue in FW 3.12 at this point I think it’s OK to close the Github issue.

If there is still an issue I think its better to have a new Github issue opened.  We are not going to debug an old FW like 3.7.2 :)

closed 2023-07-03 by gudnimg