prusa3d/Prusa-Firmware #1986: Build environment not updated to 1.0.5 in PF-build.sh

leptun, 2019-07-09:
Just like the title says, the variable was only changed in build.sh. Changing it to 1.0.5 is not enough though since @3d-gussner 's fork of PF-build-env is not updated.

deliopoulos, 2019-07-11:
You can try this [PF-build.sh](https://github.com/prusa3d/Prusa-Firmware/files/3383606/PF-build.zip) which will use my test Prusa Firmware build environment for Windows which is based on mkbel Linux64 build environment and Arduino IDE 1.8.5 portable

leptun, 2019-07-11:
I am using Linux subsystem for Windows for now with updated data in PF-build.sh for 1.0.6 and it works. Thanks for the help though. :)

3d-gussner, 2019-07-12:
I hope that Prusa will move the @mkbel `PF-build-env` repository to Prusa3d to have everything in one place. @DRracer already posted that his Arduino_Boards fork will be merged to Prusa3d. I would be fine if "my" `PF-build-env-Win` branch would move to Prusa3d to have everything together.

@leptun As the `PF-build-env-Win` branch and the ´PF-build.sh` is at this moment community driven (mostly by me) it may take a bit to follow @mkbel 's `PF-build-env` changes.

@deliopoulos I like your approach to separate the ArduinoIDE part and the Prusa specifics.
Is it okay if i use your idea in my fork?

deliopoulos, 2019-07-13:
Yes it is ok.
I can tell you that the same approach can be use both for window and linux build.

deliopoulos, 2019-07-13:
@3d-gussner Just check this [PF-build.sh](https://github.com/prusa3d/Prusa-Firmware/files/3388589/PF-build.zip)

deliopoulos, 2019-07-13:
@3d-gussner Just check this [PF-build.sh](https://github.com/prusa3d/Prusa-Firmware/files/3388593/PF-build.zip) a better approach for both Linux and windows build.

3d-gussner, 2019-07-13:
@deliopoulos Thanks...there is a small bug in it ... but easy to fix.
Your `PF-build-env` is placed under `hardware`. I prefer to place it under the `portable` folder to have this separated from other Arduino instances.

At least on Windows by default every Arduino you have installed will look in the `C:\Users\<username>\AppData\Local\Arduino 15\preferences.txt`, `C:\Users\<susername>\Documents\Arduino` and so on. Using portable prevents other Arduino instances to interfere with this one.

deliopoulos, 2019-07-13:
That's why I had also include portable folder, then this instance of Arduino is working autonomous.

DRracer, 2019-07-16:
@deliopoulos @3d-gussner could you please prepare a PR for the PF-build.sh reflecting the latest changes in build env?
Btw. AFAIK the portable subdir is one of the 3 variants of adding packages into the Arduino IDE:
- hack it directly into hardware subdir
- user profile (c:\users\....\appdata\arduino...)
- portable subdir
I personally prefer the portable subdir too.

3d-gussner, 2019-07-16:
@DRracer I am busy with the PR at this moment I try to test the changes in Linux subsystem on Windows 10 and in Windows 10 using git-bash.
I may even try to use a Linux 32-bit VM to get that also supported.

I like the @deliopoulos approach to split the `PF-build-env` into:
- Download `Arduino IDE 1.8.5` from arduino.cc . As no additional download is needed as long we stay at this version
- Download/add `Prusa Arduino_Boards` 
- Download/add `Prusa specific library changes` see https://github.com/mkbel/PF-build-env/commit/3b241c277973f5029a95fb5ca79d785802837879

I will use the portable version so users can use `PF-build.sh` to generate the firmware via script and/or use Arudino IDE 1.8.5 GUI to do so.

As many times explained I try to separate this environment from other Arduino instances/libs/boards/sketches using the portable version of it.

Status of the PR:
- Separate Arduino IDE 1.8.5 download/"install" works
- Separate Arduino_Boards download/add works just added https://github.com/DRracer/Arduino_Boards/pull/1
- Download Prusa specific library changes are ongoing ... hope to get these done today.
- Change `preferences.txt ` for GUI usage ongoing

deliopoulos, 2019-07-16:
Changes needed to PF-build.sh
[PF-build.sh.patch.zip](https://github.com/prusa3d/Prusa-Firmware/files/3397872/PF-build.sh.patch.zip)

Status of PF-build-env:
- Download Arduino IDE 1.8.5 from arduino.cc.
- Download build-env which include all changes from [mkbel/PF-build-env](https://github.com/mkbel/PF-build-env).

Advantage of build-env:
-  This version of Arduino IDE is portable (no use of user profile)
-  All changes are in the hardware subdir
-  You can use it for both git-bash and Linux subsystem on Windows 10
-  You can also use Arduino IDE GUI ( on the first run you must select "Tools -> Board: PrusaResearch EINSY RAMBo")

3d-gussner, 2019-07-17:
@DRracer @mkbel Can you please check my changes before i make the PR here
https://github.com/3d-gussner/Prusa-Firmware/tree/MK3_PF-build_106
https://github.com/3d-gussner/PF-build-env/tree/PF-build-env-v2
https://github.com/3d-gussner/Arduino_Boards/tree/Fix_maximum_firmware_flash_size
and the PR https://github.com/DRracer/Arduino_Boards/pull/1

At this moment the PF-build.sh has links to my forks/branches but i think it should be possible to get them in Prusa github, at least the Arduino_Boards.

Big thanks to @deliopoulos 

Only thing is left is a `python` check. I have on my Windows 10 and Linux Subsystem for Windows `python3` installed.
- Under Windows (git-bash) the python script `lang-check.py` works fine.
- Using Linux Subsystem i get `/usr/bin/env: ‘python’: No such file or directory` as there is just python3 installed. Linux isn't my 1st OS I am used to work with, so need bit more time for that.

deliopoulos, 2019-07-18:
@3d-gussner 
For Linux Subsystem you must install python with command `sudo apt-get install python`
If you have install different version then you have to `ln -sf /usr/bin/python3.5 /usr/bin/python`

For `PF-build.sh` you don't need `RAMBO_PLATFORM_FILE="PrusaResearchRambo/avr/platform.txt"` anymore.
`PF_BUILD_FILE_URL`  is hard-coded so no use of `BUILD_ENV`
For `arduino-builder` if you use `-hardware $ARDUINO/portable/packages`, `-tools $ARDUINO/portable/packages`, `-libraries $ARDUINO/portable/sketchbook/libraries` and user has update `arduino avr boards` via gui then updated version will be used for compile. This is not correct.

3d-gussner, 2019-07-22:
@deliopoulos 
> For Linux Subsystem you must install python with command `sudo apt-get install python`
> If you have install different version then you have to `ln -sf /usr/bin/python3.5 /usr/bin/python`

Thanks ... added to check routine

> For `PF-build.sh` you don't need `RAMBO_PLATFORM_FILE="PrusaResearchRambo/avr/platform.txt"` anymore.

Thanks, cleaned up 

> `PF_BUILD_FILE_URL` is hard-coded so no use of `BUILD_ENV`

`BUILD_ENV` is used in several locations so it is used, but I updated the `PF_BUILD_FILE_URL` to use this variable. Only downside is that the release and files have to follow this naming convention.

> For `arduino-builder` if you use `-hardware $ARDUINO/portable/packages`, `-tools $ARDUINO/portable/packages`, `-libraries $ARDUINO/portable/sketchbook/libraries` and user has update `arduino avr boards` via gui then updated version will be used for compile. This is not correct.

Right if the user updates the arv boards from 1.6.20 to 1.6.23 it will overwrite the `avr6.xn` modification and change also other libraries.

I am looking if we can add the `lcdscripts` library to the `Arduino_Boards` to prevent this. Ultimachine did it with `U8glib`. See my comments https://github.com/DRracer/Arduino_Boards/pull/1

This would make it even simpler as changes to Arduino IDE would stay in ONE location and not two or more. Easier to maintain.

3d-gussner, 2019-07-22:
Opened PR https://github.com/prusa3d/Prusa-Firmware/pull/2028

DRracer, 2019-07-25:
@3d-gussner would you mind closing this issues? Your PR was merged upstream, thank you for your hard work. We'd like to move the board definition into the official repo shortly.

closed 2019-07-25 by leptun