prusa3d/Prusa-Firmware #1988: 3.7.1 (Mk3) Temp Calibration

GWdd, 2019-07-10:
Temp Calibration has no effect on Z-Offset (no correction based on current temperature of PINDA probe.

Run Temp Calibration from LCD menus. 

Print a part using a wait for PINDA temp 35c before running 7x7 mesh bed level.  Observe layer 1.

Print a part using a wait for PINDA temp 45c before running 7x7 mesh bed level.  Observe layer 1.

The two layer 1 results are vastly different, with the second run over extruding a large degree. See attached photo.
[image]

michalxfanta, 2019-07-17:
When we tested 7x7 mesh bed level calibration, we had got better results with a fully preheated nozzle.

GWdd, 2019-08-05:
Not sure why nozzle temp would affect the results shown here -the nozzle was 215 in both cases, PINDA temp the only variable.  

Any bed warp changes over time is precisely what the mesh level should fix, which by example it isn't 
doing.  A small 10c change in PINDA temp is changing Z-offset 0.050 to 0.075 mm.  That's a fairly large change.

closed 2023-09-28 by github-actions[bot]