prusa3d/Prusa-Firmware #1993: Printer resets when MMU sensor detects no filament

jeffgros, 2019-07-11:
Printer resets when starting print and MMU sensor detects no filament.

It might be file specific. I'm using the files for the mmu2 enclosure upgrade: (https://www.prusaprinters.org/prints/3673-mmu2s-enclosure/files).

Original Prusa i3 MK3S/MK2.5S Multimaterial 2S. Firmware 3.7.2 (1.0.5) June 27, 2019 (MK3S 3.7.2-2363, MMU2S 1.0.5-297).

Steps to reproduce:
1. Have the selector at filament channel 1.
2. Ensure that selector finda does not detect filament. Having filament loaded is ok, just not to nozzle.
3. Preheat the nozzle/bed using the PETG setting.
3. Print from SD card using "prusa_mmu_enclosure_1_02mm_pet_mk3s_15h32m.gcode" (see link above).
4. After mesh bed leveling finishes, printer will reset.

In order to start the print, you'll need to load to nozzle so that the MMU filament sensor always detects filament. If you run out of filament during a print, the printer will reset, even if spool join is enabled!

uniacke1, 2019-07-30:
I think I've observed this too - my printer was resetting when it was just about to start the purge line until I disconnected the power to my MMU2.

vintagepc, 2019-10-11:
This is still present in 3.8.1 - RC, and the printer will reset even if the FINDA does not detect filament.

This will happen very often if you run a regular gcode with an MMU installed and no filament loaded. I often slice this way with large sets of prints in the same filament as it allows me to leave the filament loaded (and printer more "ready") in between prints on the MMU and only do a normal short purge line.

Printer should at least prompt for a filament change (or to select one), as would be expected in other runout conditions, full-on reset is very unexpected.

PR #2466 (2020-03-26): #2464 fix runout for unknown filament (also fixes #1993, #2301)
This fixes the printer soft-resetting if a filament runout is detected while the current filament is unknown (e.g. user loaded to nozzle, and an event caused a power cycle (power panic during MMU print), or user turned off printer and continued printing with the same filament the next day). 

Because the current filament is unknown, we cannot reliably retract it and move on to the next one if spooljoin is enabled (theoretically we could try this and let the MMU try a retract/home/next filament, but that seems risky),  nor can we m [... cut]
commit a4b8f522 (2020-03-26): Merge pull request #2466 from vintagepc/#2464-fix-runout-for-unknown-filament

#2464 fix runout for unknown filament (also fixes #1993, #2301)
closed 2020-03-26 by DRracer