prusa3d/Prusa-Firmware #1994: Can not compile current version

CdRsKuLL, 2019-07-11:
Hi, I've been trying to compile the current version. The 3.7.1 compiles and works fine. Tried the 3.7.2 and I'm getting the following error. I have copied mk3 varient over and renamed it Configuration_prusa but not changed anything else.

```
wiring.c.o (symbol from plugin): In function `__vector_23':

(.text+0x0): multiple definition of `__vector_23'

sketch\heatbed_pwm.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/users/cdrsk/onedrive - solite europe limited/personal/steves inventor/prusa/arduino-1.8.5/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board RAMBo. 
```
Same Arduino (1.8.5) compiles 3.7.1 fine, errors on this version.  Any ideas?

deliopoulos, 2019-07-11:
It's seems that you are using Windows. PF-build.sh for windows is broken. You can try custom [PF-build.sh](https://github.com/prusa3d/Prusa-Firmware/files/3383606/PF-build.zip) or patch arduino 1.8.5 with files from [PF-build-env-Win](https://github.com/deliopoulos/PF-build-env-Win)

CdRsKuLL, 2019-07-11:
Thanks deliopoulos that sorted it. 

Just for others. I clicked on the PF-build-env-Win and downloaded the files. I then copied all the files in the hardware folder into the hardware folder in C:\Program Files (x86)\Arduino\hardware  .. I then selected the new Prusa-rambo board from the arduino menu instead of the rambo board I had installed before.  :-)

closed 2019-07-11 by CdRsKuLL