prusa3d/Prusa-Firmware #1995: Sound broken again with heatbed_noise_suppression

leptun, 2019-07-13:
I think the bug came back with the introduction of timer2 as main system timer. Long beeps are now very short (like quick_feedback short).

DRracer, 2019-07-15:
@leptun indeed, you are right, we'll fix it ... or you may try too ;)

DRracer, 2019-07-23:
@leptun is this issue still present with the latest pull-requests?

leptun, 2019-07-23:
@DRracer Well... kind of. The initial issue regarding the long beeps is fixed, but there is still another issue:
With the introduction of SYSTEM_TIMER_2, all tone commands have been replaced by WRITE(BEEPER,HIGH) and don't use timer2 as the "tone" timer anymore. This can however be changed to another timer in the Tone.cpp in the arduino defaults in the new board definition. There is still one more problem though and that is the inconsistency of used timer between the two rambo boards:
Mini rambo has timer3 available only
Einsy rambo has timer5 available and maybe also timer3 if we disable the broken/unimplemented pwm support for the lcd backlight on the mk3(s).

This has to be kept in mind if we bring tone back. On a side note, I'm not sure if bringing tone back is a good thing because of the lack of flash, but also of the lack of free timers. We have to decide what is more important:

- Implementing tone
- Maybe implementing the "VGA" display mode you mentioned once.

On the Einsy maybe both could be done, but on the mini Rambo only one can be done.

As for closing this issue, do as you want. Maybe keep this open as a reminder

maximlevitsky, 2019-07-29:
On last weekend I finally had gotten to study the einsy in depth.

I attach bunch of the docs I prepared (probably still have mistakes though), later I will put them on my github or so.

In terms of timers I think that due to soft PWM used on both fan and the extruder heater,
that basically timer3,timer4, and timer5 are free to be used on Einsy, and looking at the
firmware source it looks like the only difference vs older mini RAMBO is that it uses
timer5 for motor currents, thus it should still have timer3, and timer4 available, which would
allow for both the tone() and LCD update.
Also can't LCD update be hooked to the 'big' OC2B timer routine (the ISR(TIMER2_COMPB_vect))
 
[connectors.txt](https://github.com/prusa3d/Prusa-Firmware/files/3444336/connectors.txt)
[interrupts.txt](https://github.com/prusa3d/Prusa-Firmware/files/3444337/interrupts.txt)
[pins.txt](https://github.com/prusa3d/Prusa-Firmware/files/3444338/pins.txt)
[steppers.txt](https://github.com/prusa3d/Prusa-Firmware/files/3444339/steppers.txt)
[timers.txt](https://github.com/prusa3d/Prusa-Firmware/files/3444340/timers.txt)

leptun, 2019-07-30:
@DRracer @maximlevitsky I've just tried setting the Tone timer to timer4 and it worked prefectly on the MK2.5. I'll check tomorrow on the MK3. I didn't know the fans were handled entirely in software. It's very good news. https://photos.app.goo.gl/7hm4xSXAPbuc1zJe8

About the lcd update, it's not really lcd_update() that will be called. I think it should work something like this:

All lcd print/move/anything related to graphic should change stuff in a memory buffer that is a mirror to the 80 byte buffer of the display. This way all lcd related updates only take as long as it takes to process the text on the MCU, not also on the display which is considerably slower. Let's say we use timer3 on the MK2.x and timer5 on the Einsy since it's not that hard doing this in the Prusa-firmware code, unlike in the Arduino Library. You need a dedicated timer where you push the next character on the display every 100us. Now since we use 4bit lcd mode it actually takes 100us to process one character, so in reality the speed at which you push is halved. Also every 30s or so you need to disable the VGA cycle (the one that pushes characters to the display), do the lcd_refresh() directly in case the nibbles got out of sync (because of using 4 bit mode) and then re-enable the normal VGA buffer.
You can imagine this something like a CRT display with the exception that characters stay on the screen in contrast to fading out on the CRT.

I hope to be able to implement this before the release of 3.9 and maybe have it tested, debugged if it is the case and merged by the release of 3.9. If anyone wants to join on this "project", let me know. :)

GurliGebis, 2019-07-30:
@leptun the idea with a buffer for the LCD and then splitting the render from the update logic sounds like a great way to separate it, and it should simplify things, since only ONE thing has to worry about the LCD 🙂

leptun, 2019-07-31:
For anyone interested... I don't think you know how excited I was when I saw this on the serial. Who needs a display when you have this ;)
[image]
If only the custom characters worked here...

PR #2149 (2020-02-04): PFW-1056: Re-enable Tone on timer4
This PR reenables tone functionality. This is not enough by itself though, that's why there is another PR here for the other required changes. https://github.com/prusa3d/Arduino_Boards/pull/3
Closes #2365
Closes #2229
Closes #1995
Closes #2432
closed 2020-02-04 by DRracer