prusa3d/Prusa-Firmware #1997: Options for new firmware.

gfilipek2, 2019-07-13:
Hi. I dont know if this is a proper place to write this...

1. Im using (like some people) two beds on printer (smooth and powder coated). So i have two Z settings but i must do it manually. Maybe you could put setting in menu options to chose bed and attached z offset? Now it can be done only while printing,
2. Advanced settings, why not count how much filament was used and store it on different (not global) values? So we can number spools and select it before print (or from gcode), This way we would know how many filament left :) Now we have just one global meter for the printer and cant even clear it ... dont we? :)

GurliGebis, 2019-07-13:
1 is coming in version 3.8
2 I don't think that would be coming, since it would wear out the EEPROM on the motherboard way too fast. (if you use Octoprint, there is a plugin for filament management in there, that does what you are talking about) 🙂

gfilipek2, 2019-07-14:
> 1 is coming in version 3.8

Great :)

> 2 I don't think that would be coming, since it would wear out the EEPROM on the motherboard way too fast. (if you use Octoprint, there is a plugin for filament management in there, that does what you are talking about) 🙂

Write it on SD card :)

GurliGebis, 2019-07-14:
Writing it to the SD card would solve that part of the problem, yes.
Whether or not it would make sense wrt. the small amount of flash memory left for the firmware to grow is another question :) (looking at how much work is being put into shrinking the firmware even a few bytes, it seems like they have hit the roof)

gfilipek2, 2019-07-15:
Ow.. i didnt know that J.Prusa said: "640K ought to be enough for anybody." ;)
... and now they hit the limit.

GurliGebis, 2019-07-15:
256k to be precise 🙂

closed 2019-07-15 by gfilipek2

leptun, 2019-07-15:
252kB to be the most precise. There is still the bootloader :)