prusa3d/Prusa-Firmware #1999: M600 Change filament to improve

CrNMGuy, 2019-07-14:
Hi, 

When printing and got a filament change (automatic by runout filament sensor but should be the same by in code M600 call), you unload, confirm, load, confirm color. I manually clean the "tail" extruded filament on confirmation
Then before going back XY, it extrude a little, confirmed by code inspection (see below)
Then move XY and Z 
Then heat again the nozzle (only 1-2 sec because still on the same temp for filament change)

Two thing that got your print goes bad:
1. "extrude a little" is doing a new tail like 8 mm that you may miss to catch before nozzle going back XYZ. Can be on the wrong place on the printed part
2. heating on the part after XYZ move would mostly puncture / degrade locally your print because nozzle is static in last XYZ 

**Would it me possible to move heating part on parking location before moving XYZ and perhaps the "extrude a little" is not necessary for quality printing**

```
https://github.com/prusa3d/Prusa-Firmware/blob/MK3_3.7.2/Firmware/Marlin_main.cpp

static void gcode_M600(bool automatic, float x_position, float y_position, float z_shift, float e_shift, float /*e_shift_late*/)

//Feed a little of filament to stabilize pressure
    if (!automatic)
    {
        current_position[E_AXIS] += FILAMENTCHANGE_RECFEED;
        plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS],
                current_position[E_AXIS], FILAMENTCHANGE_EXFEED, active_extruder);
    }
```
Printer : MK3S
Firmware 3.7.2

DRracer, 2019-07-16:
@CrNMGuy this ought to be fixed in 3.7.2 for the filament runout situation. The reheating part was to be made in the service position and no other heating was to be done over the printout.
The M600 handling itself is unfortunately very brittle for all the usage scenarios for now (MMU/no-MMU, automatic/gcode activated etc.). We'll look into it in the future.

CrNMGuy, 2019-07-16:
Great, thanks !

mkh595, 2019-09-27:
Hello,

I found same problem with _"extrude a little"_ when tried multi material print with one extruder (no MMU2) with tower.

There is correct change of filament with confirmation that color is OK. Then nozel moves over tower and start extruding huge amount of filament into tower without move. This cause that nozel hits this buble of filament when starts moving and break off tower from bed.

I had to commented tree lines after _; CP TOOLCHANGE LOAD_ for each filament change to avoid this issue.
If there si confirmation that filament was loaded correctly and color is OK, why is there extra feed of filament? It looks like problem is in the slicer because gcode contains commands causing issue.

PrusaSlicer 2.1.0
Printer MK3S fw 3.8.0

```G1  Y52.430 E5.0000 F1363
G1  Y109.223 E-5.0000 F682
G1 E2.0000 F2000
G1  X199.133 F2400
G4 S0
; Filament-specific end gcode

;TOOL CHANGE
T1

G1  Y109.223 
M900 K30; Filament gcode

G4 S0
; CP TOOLCHANGE LOAD
;;;G1 E18.0000 F180           it had to be commented
;;;G1  Y52.305 E63.0000 F759  it had to be commented
;;;G1  Y109.223 E9.0000 F531  it had to be commented
G1  X198.633
; CP TOOLCHANGE WIPE
G1  Y51.930 E2.5695 F800
G1  X199.133 Y52.680  E0.0404
G1  Y111.430 E2.6348 F900
```

closed 2023-09-28 by github-actions[bot]