prusa3d/Prusa-Firmware #2000: LCD out of control after MMU2 error.

antimix, 2019-07-14:
All starts when MMU2 failed, and the LCD display MMU2 Requires intervention.

After a series of operation on the MMU2 and clicks on the LCD knob,  the LCD knob become inactive, and is not possible any more to interact with the LCD and the printer.

IT HAPPENED WHILE:
- Pressing several time the buttons trying to recover the control while there is the MMU2 message.
- Just before the LCD display the MMU2 error, we are trying to navigate in the LCD menus, as example Support/Sensors.
- If the MMU2 error appears while navigating through the LCD options, AND the knob is still pressed.

shilbert01, 2019-07-28:
I am seeing identical behavior after repeated attempts to recover from MMU2S load failure. FW 1.05 and 3.7.2

gudnimg, 2023-08-02:
Closing as this is stale, we've improved the LCD behavior a lot over the years and I don't think this is an issue anymore.