prusa3d/Prusa-Firmware #2002: [3.7.2] Printer seem more sensible to power variation/outage on a UPS

forestqc69, 2019-07-15:
Printer was printing and there was bad weather outside, the was micro cuts, nut enough to have my house's lights or computer screen to go out. Printer on a UPS, but many times the printer made a beep, seemed to do a short reset (not as long as a full reset), if I remember well did go back home and continued the print.

The printer seems way more sensible to those micro-power variation, I could hear my UPS click, this never happened to me before in similar occasions.

MK3 on 3.7.2, printing from USB wire with Pronterface.

bjanders, 2019-07-19:
This may be related: Since I upgraded to 3.7.2 the printer believes there is a power failure every time I turn off the printer from the power supply button. This makes the printer beep very loudly (which is annoying), and in the printer statistics the power failure count increases every time I power off the printer.

mionut, 2019-07-31:
Hello, I have the same problem, when i simulate a power failure in ups, the printer is restarted and i resolved it by commenting `#define UVLO_SUPPORT` in `pins_Einsy_1_0.h` and `1_75mm_MK3S-EINSy10a-E3Dv6full.h` because i don't really need power panic because at my place power surges take at most 10...20 minutes.
The problem may be related to the fact that the motherboard was damaged and i bought a new one from Prusa and I received Einsy 1.1b version(the old one was 1.1a).

forestqc69, 2019-09-13:
I compiled with your suggestion @mionut, will have to test later pull pulling the cord works, need to test the UPS switching.. The settings I would like to find now is how to have the Power panic to NOT start before 50 or 250 ms so it wouldn't reset when switching on the UPS..

forestqc69, 2019-09-21:
The modification worked! I can now enjoy being on a UPS and not having the printer to reset when there's a power outage!

bjanders, 2019-10-02:
After upgrading to 3.8.0 I don't have the beeping issue when powering off anymore.

leptun, 2020-01-26:
I also had random power panics while I was using a smart power plug so that I could power on/off the printer remotely. In the end, I just stopped using that and the issue was gone. Not sure what was causing it in the first place as I was not experiencing any power cuts with other devices plugged into it.

wavexx, 2020-01-27:
> Printer was printing and there was bad weather outside, the was micro cuts, nut enough to have my house's lights or computer screen to go out. Printer on a UPS, but many times the printer made a [...]

The PowerPanic circuit is designed to trigger very fast so that we can use all the residual power of the power supply to save the print state.

What I think happens to you, which is something I also used to have in an old apartment, are small brownouts causing the powerpanic to trigger, without the PSU really being affected.

My old APC Back-UPS 500 triggered always too late because of this. often causing micro interruptions if they came just after a brownout. I had to switch to an UPS which had proper upregulation to protect sensitive equipment (the Smart-UPS line, also from APC, does this).

forestqc69, 2020-01-27:
@wavexx the problem isn't related to the ups as the problem arose with firmware upgrade. the Power panic should have a builtin timer that verify if the power is back after, let say 50-100 milliseconds, or wait a little more before going into the stopping process. This way it wouldn't continue the sequence if it was just a brownout.. But the most important thing is that this problem didn't arrived before firmware 3.7.2, so it's software related. I disabled the Power panic fonction in the firmware for some printers and remove the power panic cable for another one and the issue is solved (all printers are on ups').

wavexx, 2020-01-27:
I've just modified the code handling this and it's designed to bypass everything to start saving as soon as possible. If you wait 50-100ms after the trigger but power is not coming back you've maybe lost some precious power.

Because of this design (cutting all corners to save the current state) once the panic is triggered, a normal resume is not really possible: you need a full reset.

I'm surprised this changed with a FW release though. I've checked the difference between v3.7.1 and v3.7.2 and you see that the INT4 interrupt and/or the starting part of uvlo_ didn't change. Did you notice the difference with an earlier FW maybe?

leptun, 2020-01-27:
@wavexx It could be because of the change in bed PWM behavior that is causing these brownouts (maybe)

wavexx, 2020-01-27:
It is a good point. Maybe the slower PWM would "save you" simply because of chance? (pwm not active while brownout occurs?).

closed 2023-09-28 by github-actions[bot]