prusa3d/Prusa-Firmware #2006: MIN TEMP TO LOW For Australia

rhys100, 2019-07-15:
Hi guys, Can min temp be lowered, or at least a setting in the firmware that can be easily changed?

In Australia, at the moment it's not uncommon for temps to go under 15 degrees overnight.

We are getting failed prints because of the following.

-We start a print at 5pm
- That print pauses for a filament change as 11pm
- During early hours or morning temperature getscold and nozzle hits min temp.
- We come in at 8am and cant resume the print / change filament because a min temp is hit, which disconnects from octoprint and print is ruined.

This has happened the last 3 mornings :(

rhys100, 2019-07-15:
Just realised this has already been discussed in https://github.com/prusa3d/Prusa-Firmware/issues/1775 and being developed / talked about here https://github.com/prusa3d/Prusa-Firmware/pull/1855

michalxfanta, 2019-07-17:
I understand your problem, but the printer working temperature range is 18 °C - 38 °C

closed 2019-07-17 by michalxfanta