prusa3d/Prusa-Firmware #2007: 1st 6mm of MK2.5s bed unusable in X-Axis

bynnek, 2019-07-16:
Printer: MK2.5s
Firmware 3.7.2 
Issue:
   I've recently noticed that the 1st 6mm of the bed travel in the X-Axis is not usable.  The firmware won't let me access that area.  Nothing is binding up and the screen reads correctly that I'm at the 6mm location however it won't let me travel to 0mm.  

Confirmation:
   1) If you send the printer a G0 X0 command, it'll only move to 6mm away from actual zero (if you also send it a G0 X6, it moves to the same spot.  Sending a G0 X6 and then a G0 X0 results in no movement on the 2nd command and there is no binding anywhere or contact with the limit switch)
2) if you try to move the axis manually in the printer settings->move axis, it won't let you go lower than 6mm.  
3) if you try to print something and have the object located within the 1st 6mm of the edge, it just truncates it and won't actually print there

What I've tried to fix it:
- Factory reset printer
- Re-flash latest firmware
- Check for any interference or things that might hit the limit switch before it should be

Anyways, I'm convinced this is a firmware issue based on the 3 things I listed.  I can provide photos of what it does to the prints if necessary but I think a few minutes of trying to replicate this will show the same issue.

leptun, 2019-07-16:
Can you please try doing a XYZ calibration, but before that please post what output you get from XYZ calibration details in support menu. Maybe the origin offset is wrongly calculated.

GurliGebis, 2019-07-16:
Just to make sure, it is the MK2.5S firmware you have installed, not just the MK2.5?
If not, that might be causing some problems, since the extruder has different dimensions 🙂

bynnek, 2019-07-17:
Yep, using the MK2.5S firmware.  

Results of the XYZ Calibration:
Y distance from min:
Left: 8.36mm
Right: 7.71mm

Measured Skew 0.13 degrees

[0;0] point offset
X: -4.84mm
Y: -0.10mm

bynnek, 2019-07-17:
Previously I'm guessing my offset was 6mmin the X term (before re-running the XYZ Cal).  Now that its 4.8, I can't get below 4.8mm on travel so that makes sense.  I guess my frame got skewed...  

I'll try to re-align it all and see if that fixes it.  Seems like 4.8mm is a pretty large offset for my skew angle but I guess the Left and Right differences are pretty large.

hagan66, 2020-01-30:
I have this exact problem as have more people on Prusaprinters.org forum. I tried to print something that required the entire print area in the X-axis and it failed miserably since the nozzle will not move in to the first few mm of the left side of the bed. There is nothing blocking the extruder which can easily move to the endstop when homing.

MK2.5S

bynnek, 2020-02-01:
Nah, the firmware is fine, just the implementation of it is limiting.  Once I adjusted out my frame and got the [0;0] point offset value to nearly zero for X, I got my full travel back.  I suspect somewhere at sometime they changed something such as to prevent you from having the extruder collide with something if your frame was skewed.

yschroeder, 2020-05-03:
I have the same problem. However, for me it is "just" 4 mm. my skew is 0.03° and the printer says the axes are aligned correctly. Also, I never changed anything with the axes. I bought a preassembled MK2 and upgraded it MK2S and MK2.5S. I never touched the frame itself.

There is a workaround. You can tell PrusaSlicer the origin point and the limited bed size:

* On the printer go to `Support>XYZ cal. details`
* Press the knob twice until you are at the `Origin Point` screen
* Note down the X and Y offset (for me: `X -3.96`, `Y -0.06`)
* Change the configuration of your bed shape in PrusaSlicer
* Go to `Print Settings->General->Bed shape`
* Set the origin to the values you noted down from the printer display
* Subtract the values from the bed size and enter the new bed size (for me: `250-3.96=246.04 and 210-0.06=209.94`

Now PrusaSlicer knows about the offsets that are applied in firmware and shifts all priting moves accordingly. You will get correct warnings when you try to print outside the printable area.

However, you still lose the print area. By the way this area is lost due to the new extruder carriage design. There is a small bumper part added which presses the endstop to avoid the cable harness crashing into the main board case.

kyet, 2023-05-10:
Same here. My offset is X: -3.65, Y: 0.18 (Positive).
@yschroeder, thanks for workaround!

Prusa-Support, 2023-07-05:
Thank you for your comments and advice.
Sounds like a good trick.

I'm afraid the issue is partly due to the printed parts designs developed on MK3-series printers and backported to MK2.5.
Plus, the geometry skew can be significantly more on MK2.5 compared to MK3, due to the articulated Y-axis structure.
When the firmware compensates for a significant skew at the end of the calibration, the axis length may be reduced by a few millimeters too.

The reduced print area on MK2.5 may often be caused by mixed firmware and hardware limiting factors.

Michele Moramarco
Prusa Research

closed 2023-09-28 by github-actions[bot]