prusa3d/Prusa-Firmware #2016: MK2.5s fails self test with 3.7.2

CyberBiker, 2019-07-17:
Prusa MK2s purchased March, 2019.
Machine working well.
Machine asked for firmware update.
Updated Firmware 3.1.0 to 3.7.2
Machine would not complete self test.  Kept saying there was a wiring error in left EXT fan.
Contacted Prusa Support
If I pre-heat for PLA, the fan will come on at about 50° C for just a few seconds then turns off
Highest temp reached is about 100° C then hot end cools down.

Re-flashed machine going backwards from 3.7.2 (3.7.2, 3.7.1, 3.7.0, 3.6, 3.2 & 3.1 (which was on the machine when it was delivered))

Machine seems to be working with firmware 3.1

cimoalpacino, 2019-07-17:
Latest firmware update for MK2S is 3.2.3.
For MK2.5 is 3.7.2.
I'm not sure which one you have, because you once write MK2S and once MK2.5. It's not the same thing 😅

CyberBiker, 2019-07-17:
Sorry about that, we have MK2.5S

T4UTV1S, 2019-07-20:
Is there any update to this issue? I'm having the same thing happen to me.

CyberBiker, 2019-07-20:
Not yet.

mrmoo06, 2019-07-25:
I would check the wiring very closely.  I have seen a post where the wire on the pinda2 Y split was in the wring side of the plug that connects to the main board......
Dont know if this is of help to you but always worth going back to the very beginning and checking it all slowly step by step.

mrmoo06, 2019-07-25:
Now youve got me thinking on this, ( I only built my MK2.5s upgrade the other day)
If it fails fan check, go back and rerun but tell it the fans are not running first off, then say yes they are running.  I had this and it got me thriugh the Wizard.
HTH

tbedore, 2019-07-26:
I’m running 3.7.2 on my MK2.5s and it’s working perfectly fine.  Gets through the fan test.  Sounds like to me it might be a wiring issue on your board.  Either the plug is in the wrong slots, the rpm wire is broken on the fan, the port where the y cable is connecting to is bad.  Or the y cable itself is bad.  But I’ve had my MK2.5s since it was announced and I’ve been running 3.7.2 and it works fine.

joeynuggetz, 2019-10-03:
What if you have a 2 wire fan instead of the noctua with the 2.5 kit? I just left my hotend fan alone and didn't use the y cable or the new noctua and I'm getting the same error

tbedore, 2019-10-03:
@joeynuggetz Yeah, if you are using a regular 2 wire fan you are going to fail on the self-test every single time.  The self-test is looking for the rpm sensor to tell if the fan is spinning.  If it doesn't get rpm's it is going to think it has failed.  If you want to get past the self-test you NEED to have a 3 wire fan with the RPM sensor if you intend on continuing to use MK2.5 firmware.

joeynuggetz, 2019-10-04:
> @joeynuggetz Yeah, if you are using a regular 2 wire fan you are going to fail on the self-test every single time. The self-test is looking for the rpm sensor to tell if the fan is spinning. If it [...]

So I tried adding the 3rd tach wire to my Sunon fan and self test spins the fan but says it fails and to check wiring. Is there a current limit on the pin or something? Noctua passes test but Sunon fan which is rated higher fails. Not sure if that's what's happening.

leptun, 2019-10-04:
What rpm does the printer report with the fan on? It’s in Support -> Extruder Info (or something like that).

tbedore, 2019-10-04:
That I'm not so sure about.  I'm using a sunon fan on my 2.5 and it's working fine.  I've never used the Noctua before, but my machine passes self-test every single time

joeynuggetz, 2019-10-04:
Set extruder to 60c and fan spins but tach shows 0 in the support menu.

joeynuggetz, 2019-10-04:
> That I'm not so sure about. I'm using a sunon fan on my 2.5 and it's working fine. I've never used the Noctua before, but my machine passes self-test every single time

Which fan do you have?  Mine  is 0.72W

leptun, 2019-10-04:
That means the printer is not detecting the fan at all. Is the sunon fan wired the same? Also, is it a 4pin fan or a 3pin one? If it’s a 4pin, you might have wired the PWM input of the fan instead if TACH output.

tbedore, 2019-10-04:
Mine is the MF40100V1-1000C-G99 0.83w fan.  The MagLev one.

joeynuggetz, 2019-10-04:
HMM, interesting. Mine's a 3 wire as well. Its def wired up right. The positive and tach are on the 4 pin connector an the ground is on the 2 pin connector which is connected where the 2 wires used to be before I introduced the 3rd wire. I also dont see RPM for the part cooling fan but I never changed that and its the same that I had when I was MK2. If I let it run I eventually get a fan error and the hotend shuts down.

leptun, 2019-10-04:
4pin connector? Where have you plugged the + and TACH?
If you meant 3pin, then make sure it’s connected to X-Max endstop port.

joeynuggetz, 2019-10-04:
> 4pin connector? Where have you plugged the + and TACH?
> If you meant 3pin, then make sure it’s connected to X-Max endstop port.

Yup sorry, meant 3 pin. In X-max endstop port.  The fan I have now is the F99 so maybe not the right fan to use. I also have the 12v G99 and will give that one a go.

tbedore, 2019-10-04:
Yeah if it is spinning up that means you have the correct current going to the fan.  But the problem is the rpm sensor.  It could be the wire is damaged somewhere.  Could be the socket on the board is physically damaged.  If that port is damaged you aren't getting anywhere without directly soldering to the board.

leptun, 2019-10-04:
The endstop ports only supply 5V. If you need 12V, wire up + and - like on the old Mk2 fan.

tbedore, 2019-10-04:
Yes, needs to be 5V fan too.

joeynuggetz, 2019-10-04:
> Yes, needs to be 5V fan too.

Wow, didn't even notice the Noctua was a 5V. Luckily have both lots of fans. I had originally bought the 5v and was told I couldn't use it and searched like nuts for the 12v. I'll wire up the 5v which is the same you have and it should be sorted.

tbedore, 2019-10-04:
Yeah, people make it more confusing.  Even the MK3 a 24v system uses the same 5v fan.  The board has voltage step down converters all over the place.

closed 2023-09-28 by github-actions[bot]