prusa3d/Prusa-Firmware #2024: M350 non-functional for X/Y/Z

uniacke1, 2019-07-20:
Running 3.7.2 firmware on a Mk3.

Observed:
M350 commands are not stored/working. 
Via pronterface, I issued M350 X8 Y8 to halve the microsteps. Following this with an M503 reveals the values are still fixed at 16. (Tried both with M500 and without printer reboot in between. M350 and M503)

However, M350 E16 does, in fact, work to change the extruder microsteps from the default 32 as reported by M503.

This is confirmed by inspecting the code; the M350 handler only looks for an E-value.

Expected:
M350 X/Y/Z to function the same as M350 E

leptun, 2019-07-28:
@uniacke1 I've done the required changes for this to work. Testing tomorrow if it works. If yes then I'll make a PR.

leptun, 2019-07-29:
@uniacke1 I've tested the changes and they work, but there are some things to note here:
- On the E axis you can use 8x - 128x resolutions.
- On all other axis you can use 1x - 32x resolutions. I tried going higher on the X/Y axis, but the acceleration became erratic and at 256x when you home you get a stack overflow. Also speed is very low at these resolutions and acceleration begins to slow down a bit even at 32x.

I'm creating a PR now with the changes. If you have anything you want to change about M350, now it's the time.

uniacke1, 2019-07-29:
That sounds reasonable, thanks! I know the einsy has issues handling resolutions that are too high; I was mostly looking for some room around the default values and the ability to change without recompiling/flashing. This should handle that quite nicely.

maximlevitsky, 2019-07-29:
Tested as well. I happen to play with 0.15 nozzle recently (and it works really superb), so I tested this to see if I can extract last drops of quality from the printer this way. I tried 128 usteps on X, and 32 usteps on X,Y, and kept the default on Z. Print (0.11x benchy) succeeded, sadly without any visible increase in quality, but not worsened either. I clearly see that steppers, especially the extruder stepper sound differently, so the patch is working.

I personally won't limit the resolution on X,Y since with low enough speed it will probably work even with higher resolution, since this is non standard GCODE anyway, let the user have it.

vintagepc, 2019-08-10:
I just tried this out manually applied on top of 3.8.0-RC1 and it seems to be working well for me. Kudos!

closed 2019-11-08 by DRracer

vintagepc, 2019-11-08:
wohoo!

PR #2057 (2019-11-08): M350 for all axis - MK3
Closes #2024