prusa3d/Prusa-Firmware #2025: (3.7.2) Calibration XYZ fails way too often

Eriobis, 2019-07-20:
I just upgraded my MK2 to MK2.5S and I'm getting the calibration failed error.
After looking intop the code I've found this : 

Calibration success is driven by :
```
if (st_get_position_mm(Z_AXIS) == MESH_HOME_Z_SEARCH)
    {
```
And the get_position_mm is :
```
float st_get_position_mm(uint8_t axis)
{
float steper_position_in_steps = st_get_position(axis);
return steper_position_in_steps / cs.axis_steps_per_unit[axis];
}
```
**I am on the v3.7.2 right now
 
The axis step per unit is 3200

So the code is expecting to get : stepper_position_in_steps/3200 to equal EXACTLY 5.
If your Z axis didn't travel 16000 precise steps, it will fail. That makes no sens.
 

It should be something like :
```
float threshold = 0.10f;   
if ((st_get_position_mm(Z_AXIS) <= (MESH_HOME_Z_SEARCH + threshold)) ||
(st_get_position_mm(Z_AXIS) >= (MESH_HOME_Z_SEARCH - threshold)))
{ ...
```
This would mitigate the math error.

Also, it's bad practice to compare a float to a number, even though all the integers should be able to be represented by the float, some numbers can't so  100(float) != 100(int)

DRracer, 2019-07-23:
@Eriobis thank you for your report, can you make a pull request of your proposed changes? I agree comparing floats equal is a bad practice ;)

Eriobis, 2019-07-24:
PR Sent.

closed 2019-07-24 by Eriobis