prusa3d/Prusa-Firmware #2026: 3.7.2 cannot stop paused print

uniacke1, 2019-07-21:
FW 3.7.2 on a Mk3. 

Had a print go south on me so I paused it to assess whether it was recoverable. Decided to stop it instead.

Selecting "stop print" from the menu presents the correct screen, but selecting "yes" to confirm a stop does nothing, the printer stays on this screen. I had to hit 'no', unpause the print, and then stop it again once it had heated back up and resumed.

Expected behaviour:
Stop print when confirming, even if current state is paused.

michalxfanta, 2019-07-26:
Could you give us more information, we cannot replicate this problem.

uniacke1, 2019-07-26:
What additional info would you like? I'm not sure what I can offer besides the steps outlined above. A video of the problem?

leptun, 2019-07-26:
Does it happen on every gcode? If no, please provide the gcode.

uniacke1, 2019-07-26:
I've had it happen twice on two separate prints; let me attempt to reproduce it again tonight to ID whether this is gcode related or a general quirk.

uniacke1, 2019-07-27:
I just tried it by pausing during the attached print. When paused, I can go to "stop print" and hit "yes" as many times as I want, it does nothing. I have to hit "no", resume the print, and then stop it.

[2colorcube.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/3437778/2colorcube.gcode.zip)

uniacke1, 2019-07-27:
[Speaker_Stand_For_TV.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/3437782/Speaker_Stand_For_TV.gcode.zip)
Here's a non-MMU print it happened on as well.

closed 2023-09-28 by github-actions[bot]