prusa3d/Prusa-Firmware #2027: Error by compiling  3.7.2 Firmware

Dennis206, 2019-07-21:
I started to modify the new firmware and after I finished my mod, I wanted to compile and upload to the printer. The compiling broke up. 

I tried it another round with the original (non modifyed) firmware version 3.7.2 for I3MK3s and got the same error: 

Error from compiling with Arduino 1.8.9 :

**wiring.c.o (symbol from plugin): In function `__vector_23':

(.text+0x0): multiple definition of `__vector_23'

C:\Users\*******~1\AppData\Local\Temp\arduino_build_775742\sketch\heatbed_pwm.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board RAMBo.**

cimoalpacino, 2019-07-21:
Prusa is going away from Arduino compiler.
Arduino won't let you compile your FW due to line 836 in **marlin_main.cpp**:

`    lcd_puts_at_P(0, 0, PSTR("Debbugging build !!!"));`

Three exclamation marks (!!!) make the bootloader go into an error state (check reference: [https://forum.arduino.cc/index.php?topic=132595.0](url) or 
[https://forum.arduino.cc/index.php?topic=463814.0](url))

You have to change your code to use only one (!) or max two exlamation marks (!!):

`    lcd_puts_at_P(0, 0, PSTR("Debbugging build !"));`

marc9876, 2019-07-22:
I tried removing an exclamation mark, still didn't work

3d-gussner, 2019-07-22:
@Dennis206 Moin moin,

did you read the https://github.com/prusa3d/Prusa-Firmware/blob/MK3/README.md ?
- You use Arduino IDE 1.8.9 instead of 1.8.5
- Did you use the https://github.com/prusa3d/Prusa-Firmware/archive/v3.7.2.zip from the release tab https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.7.2 ? As the MK3 branch is a development branch and few things changed since 3.7.2 that may make compiling more difficult if you don't have the right board, libraries and build environment.

You should consider to use the https://github.com/prusa3d/Prusa-Firmware/blob/MK3/README.md#using-linux-subsystem-under-windows-10-64-bit or https://github.com/prusa3d/Prusa-Firmware/blob/MK3/README.md#using-git-bash-under-windows-10-64-bit and `PF-build.sh` as this should guaranty a know stable build environment.

DRracer, 2019-07-23:
We are not moving away from the Arduino compiler ;) but we had to create our own board definition. Still work in progress (and testing), we'll be merging shortly

marc9876, 2019-07-23:
Thanks all. Wasn't using the 3.7.2 release version. All fine with the release version. Must be a bug with the current build... First time using build.sh. works great.

3d-gussner, 2019-07-26:
@Dennis206 Is your problem solved? If so please close this issue.

ChrisVahi, 2019-07-30:
I had the same issue trying to compile the dev version Prusa-Firmware-MK3. I found [this post ](https://forum.arduino.cc/index.php?topic=328346.0) in the Arduino forum relating the error code to the timer used in the ISR for heatbed_pwm.cpp. It looks like the code should use a different timer than timer 0

https://forum.arduino.cc/index.php?topic=328346.0

Panayiotis-git, 2019-07-31:
You have to remove the previous and add a new Einsy board definition:
https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json

DRracer, 2019-07-31:
@ChrisVahi the build environment was completed last week including build instructions: https://github.com/prusa3d/Prusa-Firmware
Did you manage to compile the FW according to the latest build instructions?

ChrisVahi, 2019-07-31:
I have configured Arduino with the current build instructions but I do not yet have a functioning firmware. Steps taken:
Reinstalled Arduino 1.8.5 from 1.8.9
Changed board to PrusaResearch EINSEY RAMBo
Modified compiler flags in C:\Program Files (x86)\Arduino\hardware\arduino\avr\platform.txt -- this may be where things went wrong, the line in my file was different than the example, mine is now:
compiler.c.elf.flags={compiler.warning_flags} -Wl,-u,vfprintf -lprintf_flt -lm, -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections
Copied 1_75mm_MK3-EINSy10a-E3Dv6full as Configuration_prusa.h, moved to Firmware folder
Changed LANG_MODE to 0 in Firmware.config
Made edits to extruder steps, direction and thermistor settings for my setup
Compile and load with Octoprint plugin Firmware Updater.

After uploading firmware 3.7.2 I get an "upload successful" message but the printer will not connect over serial and the LCD display hangs on the boot message indefinitely. 
I was able to upload the precompiled firmware from prusa3d.com/drivers and my compilation of 3.7.1 which I am back to now using the same upload process.

I remember having issues in the past with the external flash memory chip, could that still be a problem? I previously commented out a line of code specifying that chip's use, I'll dig through my old posts to find the fix for that later.

In the mean time I am disabling the firmware check gcode in PrusaSlicer

3d-gussner, 2019-08-01:
@ChrisVahi As Arduino IDE in normal mode uses shared folders the different version or leftovers from previous updates can interfere.
Default folders:
- Shared folder `C:\Users\<username>\AppData\Local\Arduino15`
- Sketchbook `C:\Users\<username>\Documents\Arduino`

Best way to have a "known" stable Prusa Firmware build environment is to use Arduino IDE in portable mode, see https://www.arduino.cc/en/Guide/PortableIDE and even change the sketchbook folder there.

For the `PF-build.sh` script I made a Windows build environment that you can use even without `Linux subsystem` or `git-bash` by just downloading it from here https://github.com/3d-gussner/PF-build-env/releases/tag/Win-1.0.2 . Extract it and you should be able to compile your code with that.

At this moment please **never ever** update the `Arduino AVR Boards` in the `Boards Manager` they need to stay at version 1.6.20.

If you want to compile FW3.7.2 please use source code from the release https://github.com/prusa3d/Prusa-Firmware/archive/v3.7.2.zip you might also need to use Rambo boards definition 1.0.0 and not the newest.

I am busy with a new boards definition for the `Prusa Research AVR Boards` so hopefully in the future there will be no need/less manually steps needed to compile the firmware.

If you still struggle to get the firmware compiled give these a try:
https://github.com/prusa3d/Prusa-Firmware/tree/v3.7.2#using-linux-subsystem-under-windows-10-64-bit
https://github.com/prusa3d/Prusa-Firmware/tree/v3.7.2#using-git-bash-under-windows-10-64-bit

Hope that helps.

closed 2023-09-28 by github-actions[bot]