prusa3d/Prusa-Firmware #2033: Color change returns to print in a location where new color is not needed/wanted?

stevenminnick, 2019-07-23:
I posted this in the Prusa forums and was told to post this here instead.
https://forum.prusaprinters.org/forum/english-forum-general-discussion-announcements-and-releases/prusaslicer-color-change-returns-to-poor-location-maybe-its-printer-firmware/

I've noticed some weird behavior and wasn't sure if this is on purpose.  

This is what I want to print
[image]

All pretty

But when the first color finishes it finishes at a corner of the print, lets assume where I draw the arrow.

[image]

I do the color change.  And the print head moves back to the print, but instead of going where the new color begins it goes back to where the old color left off.  So it deposits some new color here, and then moves to where the new color starts. 

[image]

So the print comes out like this.  I can remove that little dot with tweezers.  It's not that big of a problem.

[image]

Just seemed like an oversight.  Or a bug.  Why would it go right to the previous end point, deposit color, then move to where the new color is supposed to start.

Maybe I didn't notice it on previous firmware, but I've been noticing it lately once I swapped over to Prusa Slicer 2.0 (now that color change is built in) and the latest couple of firmware releases.

leptun, 2019-07-23:
First of all, this is not firmware related. It should be posted on the PrusaSlicer issue tracker: https://github.com/prusa3d/PrusaSlicer/issues

Now for the issue itself. When you set colorchange in prusaSlicer it issues an M600 imediately after it finishes the previous layer. After the change it will return exactly where it left off. It can’t go elsewhere, that’s not how this command works. A sollution to this problem would to insert the M600 in a better possition, preferably after it moves to the new color position. This would be at the begining of the next color layer, not at the end of the previous layer.

stevenminnick, 2019-07-23:
Interesting.  Thanks.  I'll post this over there. 

I used to do lots of color changes with Prusa Control and never noticed this.  I just updated to my Mk3S and the latest Prusa Slicer 2.0 and noticed it and didn't know if it was a new bug or maybe it's always been there and the new models I had just happened to show this behavior.

leptun, 2019-07-28:
Would you mind closing this issue since you have posted on the PrusaSlicer repo?

stevenminnick, 2019-07-29:
Sure, I guess.

closed 2019-07-29 by stevenminnick