prusa3d/Prusa-Firmware #2036: Unable to change sheet selection away from second choice

nebbywan, 2019-07-23:
Not sure when this started yet, but with commit 717b710abe5864a446e44e713d01958709cbc05a when the root menu sheet selection method is used, I am unable to change to anything other than the second sheet in the list. If I have the first or third sheet selected, it will change to the second, but further clicking on it remains on the second selection. The only way I am able to change to any other sheet is through the Settings->HW Setup menus. Also, I am unsure if this is related, the "Select" option is missing the the third sheet's menu in the HW Setup screen, and going into the third sheet's screen sets the third sheet as selected.

nebbywan, 2019-07-24:
Root menu issue fixed with fb35009628773968f5dbb706de1cd5656b9c07f2. Thanks @NotaRobotexe. Should I branch off the HW Setup problem into a new issue?

DRracer, 2019-07-25:
@nebbywan there might be some issues with non-calibrated sheets. These sheets should not be offered in the selection menu item. Please wait a few more days - we have a testing session underway, more issues may occur.

maximlevitsky, 2019-07-29:
Note that I noticed that you want to disallow the user to use the non calibrated sheets. This is a good idea overall, but please allow some kind of override, for the cases when the user for example cleared the EEPROM and doesn't want to calibrate each sheet again.
Currently this is possible albit tedious but selecting Z calibration from the sheet menu and then canceling it, and then setting up the live Z value manually from the settings menu.

DRracer, 2019-07-31:
@maximlevitsky well, you can always patch the source code and disable the checks, but that's probably not what you asked for. For now we'd like to have this feature as **user-safe** (read user-resistant :) ) as possible, so no relaxation of checks. But we may change that after collecting feedback from broader range of users.

maximlevitsky, 2019-07-31:
@DRracer  

"Programming today is a race between software engineers striving to build bigger and better idiot-proof programs, and the Universe trying to produce bigger and better idiots. So far, the Universe is winning"

Which reminds me that I managed to create not just one but two holes (*) in the pei sheet by using that Z axis move on long press during the print. I guess I am one of those better idiots :-)

First time it happened after I noticed that it would show me that Z values during the mesh bed leveling. Once it worked (I haven't rotated the knob for sure) and next time (I also didn't rotate it), it send Z straight to -10.0... Ouch.

The other time it happened when I raised the nozzle this way when printer was preheating. It done preheating while the raise was still happening and surely enough drove into the bed when homing.

But seriously, there should be a way to set live Z for all the sheets without doing calibration. Please don't remove the workaround I found.

DRracer, 2019-08-02:
@maximlevitsky 

> So far, the Universe is winning
exactly :)

Ok, we won't make any further restrictions on the sheets now, but please note there might be some changes after we gather more user input on this new functionality.

closed 2023-10-03 by github-actions[bot]