prusa3d/Prusa-Firmware #2039: LCD bad communication

GibranValle, 2019-07-24:
When I upload the 3.7.2 firmware, my LCD displays a 0 and a checkmark, after pressing the knub, the LCD displays symbols and restarts. 

However, if I upload the 3.4 firmware or lower everything works fine, I selected the MK2.5 variant. 

Video: https://drive.google.com/open?id=1izq9CRZq8MDwVZsC5xmg4nz7AK3_O9Xs

leptun, 2019-07-24:
Did you compile the firmware yourself? If yes, how?

GibranValle, 2019-07-24:
I downloaded the repository from https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.7.2, then I chose the MK2.5 variant, and replace the file name with Configuration_prusa.h, the arduino IDE did not find any errors.
`<iframe src="https://drive.google.com/file/d/1LKMz2-OuRh5A99IoVc56NAHJUBsDEj5s/preview" width="640" height="480"></iframe>`

leptun, 2019-07-24:
I suppose you put Configuration_prusa.h in the Firmware folder, not the variants folder.

Also you didn’t set LANG_MODE to 0 in config.h

GibranValle, 2019-07-24:
Exactly, I've always putted this file inside the firmware folder, I did not changed any other file, because in older iteration like 3.4 this procedure was enough to flash the printer.

leptun, 2019-07-24:
At the end of the file you can change it. It is for multiple language support. It disables it. What you were seeing on the lcd was the printer trying to find the other languages that don’t exist. You can’t compile with multiple languages on Arduino unfortunately. 

I suggest you read the new README in github and see how the new compile method is.

GibranValle, 2019-07-24:
Thank you, sir I will check it out.

leptun, 2019-07-24:
@GibranValle Is the issue fixed? If yes, please close the issue.

GibranValle, 2019-07-25:
After the change in the LANG_MODE to 0 in config.h the problem was fixed. 
Thank you so much.

closed 2019-07-25 by GibranValle