prusa3d/Prusa-Firmware #2045: M300 reset

bmuessig, 2019-07-25:
Trying the following command will emit an approx. 4s tone (which the buzzer struggles to do) and in turn crash and reset the printer:
`M300 P10000 S7000`

I suspect that there could either be an integer overrun or - much more likely - a WDT reset due to the too long tone. Much larger time values only yield a short tone.

leptun, 2019-07-25:
@bmuessig Is this happening in one of the releases or in the latest Mk3 branch?

bmuessig, 2019-07-25:
@leptun This is happening in a release 3.3.0. I unfortunately cannot test any more recent version.

bmuessig, 2019-07-25:
I guess that this has not been fixed in the meantime though.

Panayiotis-git, 2019-07-25:
The issue exists also to the latest version

leptun, 2019-07-25:
Found the problem. I'll test tomorrow and maybe even make a PR.

techfreek, 2021-01-10:
Still seeing this on 3.9.1 (albeit with a 5 second beep, not a 4 second beep)

Jaxx005, 2021-11-28:
Latest release FW (V3.10.0) on MK3S has the problem - it seems related to the P period:

M300 P2000 S0  ; silence for 2sec as expected
M300 P5000 S0  ; causes printer restart!

Pronterface test confirms this line as problematic - note the ERROR after reboot (??).
Here's the Pronterface USB/serial output for M300 with P2000 and then P5000:

_**>>>M300 P2000 S0**_
SENDING:M300 P2000 S0
**_>>>M300 P5000 S0_**
SENDING:M300 P5000 S0
start
echo: 3.10.0-4481
echo: Last Updated: May  6 2021 13:52:17 | Author: (none, default config)
Compiled: May  6 2021
echo: Free Memory: 2054  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Extruder fan type:
NOCTUA
CrashDetect ENABLED!
E-motor current scaling enabled
E-motor current scaling enabled
E-motor current scaling enabled
E-motor current scaling enabled
E-motor current scaling enabled
E-motor current scaling enabled
FSensor ENABLED (sensor board revision: 0.4 or newer)
E-motor current scaling enabled
Sending 0xFF
**_Error:volume.init failed
[ERROR] Error:volume.init failed_**

LCD status changed
MMU not responding - DISABLED

NOTES:  
Using the G4 (DWELL) instead of "M300 P12345 S0" appears to function OK.
Example (tested OK):

   G4 P5000 ; pause for 5sec, eg. between sounds

TTFN Cheers

techfreek, 2021-12-05:
I kinda wonder if a "quick hacky fix" for this is just to limit the duration of the M300 to be less than the max timeout of the watchdog/whatever is detecting the hang

bmuessig, 2021-12-05:
M300 is implemented here: https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/Marlin_main.cpp#L7793

I would however recommend changing the `Sound_MakeCustom` method in general and to change the delay to be executed in a loop, where the WDT is reset every iteration. This method can be found here: https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/sound.cpp#L66

bmuessig, 2021-12-05:
On second thought, it would be a wise idea to implement the duration of the beep similar to how dwell is handled:
```
if(codenum != 0) LCD_MESSAGERPGM(_n("Delay..."));
st_synchronize();
codenum += _millis();  // keep track of when we started waiting
previous_millis_cmd = _millis();
while(_millis() < codenum) {
  manage_heater();
  manage_inactivity();
  lcd_update(0); // NOTE: It would make sense to remove this for the tone
}
```
This would ensure that all critical tasks are still being taken care of during long beeps and it would act similar to a regular delay, but with the desired sound. Generally, it would make sense to only run this loop for sounds that exceed a certain threshold, say 500ms.
I generally would not pause the regular updating of the printer for extended periods of time, so this compromise should make sure that the printer keeps operating safely. As the actual sound generation is taken care of in a timer, added delays caused by the continued updating should not interfer with the sound generation itself and they should at worst add a marginal bit of delay to a command, for which slight delay inaccuracy should not matter at all.

Jaxx005, 2021-12-07:
That’s what I was wondering. Yes a bit ‘hacky’ but better than just a ‘crash’.  As long as it’s documented too 😊

bmuessig, 2021-12-07:
> That’s what I was wondering. Yes a bit ‘hacky’ but better than just a ‘crash’. As long as it’s documented too
For a quick fix, it would be a good idea to implement the maximum time-limiting in `Firmware/sound.cpp/Sound_MakeCustom()`.
This would cause other usages of the custom sound method to also respect the WDT. It's also a good idea to add a print statement that logs if a command used a too long print.

Generally, I'd recommend fixing it correctly by implementing the beep 'dwell-style', which should only take a couple of minutes to do and would be IMO the more suitable solution, as it ensures that the general updating is not paused during beeps, as it is currently.

bmuessig, 2021-12-07:
On third glance, how about simply using `delay_keep_alive` in `Sound_MakeCustom` instead of `_delay`.
I believe that is the cleanest and at the same time simplest solution that should avoid any WDT reset.
Additionally, we can still limit the maximum duration of M300 to something reasonable (e.g. 10s).

leptun, 2021-12-07:
Using delay_keep_alive would stop the crashing, but any beep from the UI while M300 is executing would stop the M300 beep, but keep waiting without any beeping.

bmuessig, 2021-12-07:
> Using delay_keep_alive would stop the crashing, but any beep from the UI while M300 is executing would stop the M300 beep, but keep waiting without any beeping.

Good point!
Where is `delay_keep_alive` defined? What components are being updated by it?
Generally, I'd suggest to only use `delay_keep_alive` for sufficiently long delays (e.g. > 1s).

I see two options to avoid using the beeper:

1. For long delays, wait in e.g. 50-100ms increments and restore the sound configuration in every iteration if it changed
2. Or, introduce a guard flag that ensures that the `Sound_MakeCustom` or its friends can only be called once and aren't re-entrant

bmuessig, 2021-12-07:
Alright, I've prepared a pull request that implements my first proposed solution, which modifies `Sound_MakeCustom` to use a delay method that suits the delay:

For delays...
* below 1.1s, use the regular delay method
* above, use `delay_keep_alive_headless`, a modified version of `delay_keep_alive` that does not update the LCD

This should be able to take care of all WDT reset problems concerning sound.

bmuessig, 2021-12-07:
Without LCD updates (which also take care of buttons and menus), there should be no possible collisions between sounds.
What do you think, @leptun?

3d-gussner, 2025-02-11:
@bmuessig I have been testing the limit and `M300 P4500` was kind of a limit with the MK404 SIM. But to be safe I have chosen `4000`ms as limit.
 
Also Vanilla Marlin limits the `M300` to 5s see https://github.com/MarlinFirmware/Marlin/blob/4d0c6841f8052c1a44be3c697ae6516a877313e5/Marlin/src/gcode/lcd/M300.cpp#L54

bmuessig, 2025-02-14:
> [@bmuessig](https://github.com/bmuessig) I have been testing the limit and `M300 P4500` was kind of a limit with the MK404 SIM. But to be safe I have chosen `3500`ms as limit.
> 
> Also Vanilla Marlin limits the `M300` to 5s see https://github.com/MarlinFirmware/Marlin/blob/4d0c6841f8052c1a44be3c697ae6516a877313e5/Marlin/src/gcode/lcd/M300.cpp#L54

Great, thank you for taking care of it!

3d-gussner, 2025-02-17:
@bmuessig Thanks for pointing out the issue and being so persistent to get it finally done after nearly six years.
Sorry it took so long.

bmuessig, 2025-02-17:
> [@bmuessig](https://github.com/bmuessig) Thanks for pointing out the issue and being so persistent to get it finally done after nearly six years. Sorry it took so long.

Better late than never :)