prusa3d/Prusa-Firmware #2049: Flashing compiled firmware does not work

Demolux, 2019-07-26:
Hi!

Today I tried to flash the firmware on my two MK3S printers but the flashing process does not work fully: When there is a flashing success message either via Arduino or PrusaSlicer, the printer still hangs in "Flashing Firmware - do not disconnect" (Arduino) or "Verifying ... 100%". What I tried until now:

- reinstalling Arduino 1.8.5
- using another PC
- using builds 3.7.1, 3.7.2 and the most recent development version

I strongly guess this has something to do with the new Board Manager as this was the only thing updated in between todays tries and my last successful build&flash about 2 months ago.

Can someone propose a solution?

Thanks in advance

leptun, 2019-07-26:
Could you please make a video of the process and also post the avrdude log and the compiled hex so I can test it.

Demolux, 2019-07-26:
Alright, here is a quick and dirty cut/video documention of the process, the hex-file and the avr-dude-logfile. All resources can be viewed via this link to my onedrive. 
The log and the hex are here on github too (small enough).
[avrdude log.txt](https://github.com/prusa3d/Prusa-Firmware/files/3436070/avrdude.log.txt)

[firmware-3.7.1-test-forgithub.hex.zip](https://github.com/prusa3d/Prusa-Firmware/files/3436072/firmware-3.7.1-test-forgithub.hex.zip)

leptun, 2019-07-26:
Check out #2039 . It’s the exact problem.

Demolux, 2019-07-26:
> 
> 
> Check out #2039 . It’s the exact problem.

Unfortunatly it is not. Lang mode is set to 0 as well as every other instruction has been followed given by the windows flashing manual on the main github page.
This problem also exists with the other printer, also an MK3S.

leptun, 2019-07-26:
I'm trying to download the video, but my internet is very slow. You said something about the new board manager. In the latest Mk3 branch we have moved to another board definition. It has some changes for the latest improvements that couldn't be accomplished otherwise. The thing is that these changes are only compatible for the upcoming 3.8.0-RC1 and shouldn't be used on older versions of the source. So you could try to compile the 3.7.2 fw with the rambo definition or the latest branch with the new board definition. If you use the wrong board definition the firmware will get stuck since it the main system timer interrupt was changed in the new beta and you would essentially have no system timer running on the old versions. (ie millis, micros and delay would get stuck or report wrong values).

I'll watch the video closely once it downloads and see from there if what I said above doesn't work.

leptun, 2019-07-26:
I think this was the old boards link if you don't have it anymore https://raw.githubusercontent.com/ultimachine/ArduinoAddons/master/package_ultimachine_index.json

Demolux, 2019-07-26:
Thank you very much for the old boards link. I tried two things now:

1. Flash printer 1 with FW 3.7.1 via the PF-build script in a sub-ubuntu-system. SUCCESS.
2. Flash printer 2 with FW 3.7.1 via the Arduino method but with the old board definition. SUCCESS.

So I guess something causes trouble because of the new board definition. Tell me if you need further resources, I have no problem delivering them.

leptun, 2019-07-26:
Good to hear it compiles correctly. Things will get sorted out once 3.8.0 is released. Till 3.8.0 is released the readme.md file should be updated to include instructions for older firmwares as to not cause confusion anymore.

closed 2019-07-26 by Demolux