prusa3d/Prusa-Firmware #2051: Proposal: Save more internal flash

3d-gussner, 2019-07-26:
Hi,
I already made a comment how we could save more internal flash here https://github.com/prusa3d/Prusa-Firmware/pull/1644#issue-261871333

But nothing happend, so I try it again.

Possible improvement to save more internal flash on MK3/MK3S EINSy:

We are hitting the EINSY resource limits quite often, so I was thinking about a different approach to use the internal and external flash for the languages.
- Now all the english messages are stored in the flash
- for the additional language translation a reserved space for secondary language (12032 bytes) of the internal flash is used. When a user chooses another language translated messages will be "offset" to the reserved 2nd language location. As most of the english internal flash space is then unused, it is quite waste of resources.

- Wouldn't it be better to have just **one** reserved 'language space' in the internal flash?
  - Few messages have to stay in the internal flash to be able to switch between the languages
  - All other english messages are stored in the reserved 'language space' and used as it is done now with the other languages
  - As soon  another language is chosen by the user, the language from the external flash is copied to the 'language space' replacing all existing messages.
     - All languages **including** english need to be flashed to the external flash during firmware update
  - this way i guess we could save 8-10kb of internal space

This will work on the MK3/MK3S EINSy with it's external flash but for the MK2.5/MK2.5S miniRAMBo we still would need the 2nd language internal flash space reserved.

But as the MK2.5/MK2.5S miniRAMBo firmware is way smaller than the MK3/MK3S (no Trinamic, no Power Panic, and other things) it isn't an issue here. I have on my MK2 4 languages and no resource issues.

maximlevitsky, 2019-07-29:
I think that this is a very good idea.

vertigo235, 2019-07-29:
I think for the EINSY it makes sense but then with the mini Rambo you would end up needing to have a hex for every language however.

3d-gussner, 2019-07-30:
> for the EINSY it makes sense but then with the mini Rambo you would end up needing to have a hex for every language however.

@vertigo235 We already have 6x hex files for the miniRAMBo MK2.5 firmware. For each language one.

As i wrote above:
You keep this secondary space for the MK2.5/s as this firmware doesn't reach the resource limits like the EINSY with the TMC drivers, PowerPanic, 2nd fan tacho... does. And as the miniRAMBo doesn't have an extra external flash as the EINSY you gonna need some flash space for that.

I had in the past some strange issues when I hit the limits around 95%-97% with my custom MK2 firmware adding additional features (FRS) and languages. So I decided to cut these down to max 4. This way the MK2/s inlc. MMU1 firmware version could also have some verbose/debugging features activated, which were very helpful.

It is always a "pain in the bu77" try to debug issues in the firmware if you cannot activate verbose/debug functions with multi languages enabled, as it quite often runs out of ressources.
Also some verbose functions have been un-commented due to resource savings.

If you look in the MK2 branch, the firmware has 6 languages stored in flash and this have been "downgraded" to two languages in the MK2.5 versions.
- We have 1x hex file for MK2 firmware including 6 languages
- and now we have 6x MK2.5 hex files each containing two english and the other one.

If someone wants to go crazy he could add the early (EINSY daughterboard)[https://d17kynu4zpq5hy.cloudfront.net/igi/prusa3d/UOC2SvXnj1Cwn3O6.full] to his miniRAMBo to have the same external flash. It should work maybe pinout has to be bit different, didn't dig too much in to that.

vertigo235, 2019-07-30:
Fair enough, I still like your idea even if the mk2/2.5 has to have more hex files.

michalxfanta, 2019-07-30:
Great idea, we are certainly considering this option, but so far we have tried to save flash by less invasive code modifications.

rdragonrydr, 2019-08-01:
Why does the mk3 branch even copy extra languages to the internal flash? If it's already stored on the extflash on the mk3, can't it just be read from there?

...For that matter, how does the flash update process _get_ them there in the first place? Is there a stub and a special host program (beyond that needed to copy the data to the cpu's internal flash) that copies them over?

3d-gussner, 2019-08-01:
> Why does the mk3 branch even copy extra languages to the internal flash? If it's already stored on the extflash on the mk3, can't it just be read from there?
> 

I am not sure if you run into LCD performance issues reading it from external flash, but we see lot of pull requests to this matter so it may be easier to tune the code for internal flash than external.

> ...For that matter, how does the flash update process _get_ them there in the first place? Is there a stub and a special host program (beyond that needed to copy the data to the cpu's internal flash) that copies them over?

Prusa Slicer has it own upload procedure to flash the firmware and than upload the language parts into the external flash. If you want to know more about that you should look there.

3d-gussner, 2020-03-26:
Too long open ...

3d-gussner, 2020-03-26:
Reopen it ...

closed 2023-10-03 by github-actions[bot]