prusa3d/Prusa-Firmware #2054: LCD show MMU error 0

shilbert01, 2019-07-28:
During a print the LCD begins showing weird output. I can decipher MMU and error 0 (actually Fehler 0 - in German).

The print (two color sheep) seems to continue just fine.

shilbert01, 2019-07-28:
A few minutes later the LCD stopped scrolling content left and right and I can now see that this is not an error but show the number of load failures for the MMU. So in other words it is just the normal display output but the content gets scrolled off-screen and moves around for some time.

leptun, 2019-07-28:
This sounds like problems in translations where the displayed text is too long and overflows on the next line. https://photos.app.goo.gl/N8AnJcfDX6pps83K7
Take a look at another example in italian. Is it simmilar?

leptun, 2019-07-28:
Also what firmware version are you using? (both printer and mmu). And what hardware? (S or not)

shilbert01, 2019-07-29:
Firmware is 3.7.2-2363 on Mk3S and  1.0.5-300 on MMU2S. Language is German.
I will upload a photo later today.

closed 2023-10-03 by github-actions[bot]