prusa3d/Prusa-Firmware #2055: M0/M1 Pause not displaying text/waiting for input

msbeck, 2019-07-29:
> M1 Hello World Waits until user presses the knob. "Hello World" message is shown as LCD status message.

https://github.com/prusa3d/Prusa-Firmware/wiki/Supported-G-codes

M0 and M1 do not prompt the user with text when pausing. They also do not wait for user input to resume.

```
G1 X10.000 Y200.000 E0
M300 S440 P200
M300 S660 P250
M300 S880 P300
M1 Click to continue 
M108 
M105; return to current temp
```
I would expect the printer to move out of the way, play the beeps, pause, display "Click to Continue" and wait until the LCD button is clicked. Instead, it just pauses, does now show message, and does not resume when button is clicked (standard Prusa menu is up).

The only way to restart is to click Restart in Octoprint, which is not ideal when the pause is one that requires user interaction AT the printer.

This is experienced while printing through OctoPrint on an MK3s. It doesn't work embedded in GCODE nor through the terminal in Octoprint.

Is there another way to go about pausing and waiting for user input?

Thank you

michalxfanta, 2019-08-05:
We will fix this bug in some of the next releases, thanks for reporting.

3d-gussner, 2020-02-03:
@msbeck I tried it myself on newest FW and it works with following gcode:
```
G28 W
G80
G1 X10.000 Y200.000 E0
M300 S440 P200
M300 S660 P250
M300 S880 P300
M1 Click to continue
G1 X200.000 Y10.000 E0
;M108 
;M105; return to current temp
```

As `M108` isn't implemented in Prusa firmware see your link above I guess it fails because of the M108 which isn't supported.

Can you please fill in the things below if the problem still exist?

Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]

**Describe the bug**
  A clear and concise description of what the bug is.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

Panayiotis-git, 2020-02-03:
It is working for me also.

Although, the current implementation is unsafe at least.

Try to use reserved words like 'PRUSA' or any of 'M' or 'G' in your message and the command does other things...

Panayiotis-git, 2020-02-03:
Check also https://github.com/prusa3d/Prusa-Firmware/pull/1900

Till now, the only reliable implementation is what is proposed at the https://github.com/prusa3d/Prusa-Firmware/pull/2380, to enclose the text into double quotes to all these custom message commands.

3d-gussner, 2020-02-04:
@Panayiotis-git Thanks for the test and update. I can confirm that `M0/M1` with a message starting with a `G` doesn't work. `M0 More Coffee` on the other side works.

sarf2k4, 2020-10-20:
I don't see anything on the lcd when I issued `M0/M1 S20 wait me` from octoprint terminal, I'm using the latest octoprint and mk3s firmware as of this writing

closed 2023-10-03 by github-actions[bot]