prusa3d/Prusa-Firmware #2063: Setup Wizard potential bed-scratch issue?

rdragonrydr, 2019-08-01:
In at minimum the 3.7.1 firmware currently on my printer (I plan to update it once my current run is completed), the initial setup wizard has the potential to scratch the heater board on new printers.

The problem arises after the printer tries to find the initial mesh calibration, and when we are asked to put a piece of paper under the extruder to find the proper PINDA leveling probe height. If this height is too high for any reason, and the test must be canceled when it grabs the paper, the printer resets right back into the mesh calibration mode (also preventing the user from going to the "move axis" menu). The first move it makes is in the X, _not_ the Z direction. It moves the extruder back to the left, THEN up. This left a faint scuff in the surface of the heater (as did/do attempts to move the extruder so that the PINDA can be adjusted).

Can the printer run a few millimeters up in the Z direction before doing any travel on resume/start of this test? Or perhaps an adjust-Z window (not for the first layer height, but to get the extruder to a more manageable position) so that the Z-axis can be moved -- without causing skew due to uneven manual movement of the screw rods -- before resume or to better adjust the probe?
-------
There's also a weird issue where, after printing and if running a cold start later, the bed position is at the front of the printer, and +Y is also to the front of the printer. This means that "move axis" cannot control the Y-axis more than a few millimeters before hitting a software stop (at "0" due to not knowing the origin yet) or when it hits the physical front of the printer. This is kinda fixable by running the "go home" command (which strangely _can_ move the bed back), but is there any way to just check the motor driver for when it has _actually_ hit an endstop on move (well, the driver detecting that the motor has stalled, or however that works) and halt that way instead?

Minor quibbles aside, though, I could not be happier with my new printer. Thanks, guys!

unixunion, 2019-08-03:
I noticed this exact same behaviour and wanted to come here and report it. +1

michalxfanta, 2019-08-05:
Thank you for your feedback, we will look into them.

RV12vaitor, 2019-11-12:
Having a similar issue with 3.8.1 on a MK2.5S MM2S, ran the calibration wizard and it drives the nozzle into the bed at the home position before the hotend/heated bed test is passed. The bad part is as soon as the bed test completes a 10mm X-axis and a 10mm Y-axis movement, along with an X/Y diagonal is made with the nozzle in the bed. I now have a nice shinny copper etching in the bed (.04mm Olsson ruby nozzle). As the Z-axis calibration is undefined at that point, perhaps it would be better to move the X-axis to the top end stops before moving the X or Y axis at the start of searching for the bed calibration points.

rdragonrydr, 2020-06-09:
Is this addressed in 3.9's unknown Z solution? Or is this still an issue?

I saw that there was a note about detecting a crash, but with two stepper motors I can see it possible that in _un-levels_ the bed by having the print head (for example) on the left: The left motor stalls, but if the right keeps spinning, it can desync the two rods that move the z-axis. Is there anything to stop them fast enough or revert any additional movement?

rdragonrydr, 2023-10-01:
Not stale.

3d-gussner, 2023-10-03:
@rdragonrydr I have tried yesterday few time to reproduce the issue and from what I can see the Z axis moves always first up before it is homing/moving the x and y axis in the current firmware version.
Please verify if this issue is still valid and provide some more information about firmware version and a step by step guide.

rdragonrydr, 2023-10-07:
Thank you, @3d-gussner, for pointing that out. It used to be that after it finds the four anchor points for sensing the bed, where you have to make sure it doesn't grab the paper, it'd make a horizontal move before lifting. Apparently that **had** been changed since I opened this, but for whatever reason it never made it back here and I never heard about it from the changelogs.

Also, running the wizard did inform me that the Y-belt was a bit loose, so that was nice (though less fun to have to fix mid-process).

I believe this issue can now be closed.

3d-gussner, 2023-10-07:
Thanks for the update and confirm that it has been solved.

closed 2023-10-08 by rdragonrydr