prusa3d/Prusa-Firmware #2066: Feature-Request: Preheat Profile for "low temp" BDP (extrudr Pearl or extrudr Flax): 180 °C / 55 °C

mavelli, 2019-08-01:
background: 
just got my first spool of extrudr's BDP Pearl filament. it prints at lower temp than PLA and i am still optimizing my filament profile for prusa-slicer. 

setup / problem:
on my mk3s with 3.7.2-2363, i have in the [preheat] menu no option for lower temps than PLA (215 °C / 60 °C), quite some manual work is involved when switching filament and then wanting to extrude a bit more as the logic requires you to preheat the nozzle with the preheat menu if you want to move the extruder position - makes sense. but the BDP filament material is rated to be used optimally at 170-200 °C (prints great at 176-191 °C from what i know already) so i would want to preheat the system also accordingly.

feature request addressing the problem: 
Preheat Profile "BDP" with 180 °C / 55 °C temps

mavelli, 2019-08-03:
knowing the filament now a bit better, i titrated it to 186 °C to be ideal for printing - so preheating i think would be too - 186 °C / 55 °C

michalxfanta, 2019-08-05:
We will consider new preheat preset, thanks for the idea.

closed 2023-10-03 by github-actions[bot]