prusa3d/Prusa-Firmware #2067: static assertion failed: Z-range too high

TigerClaw33, 2019-08-03:
Hi, 

I am using a mod but cannot change the max. Z-value from 210 to 320. 

Error: Marlin_main.cpp:3017: error: static assertion failed: Z-range too high, change the T type from uint8_t to uint16_t

Vogs27, 2019-09-17:
Same problem too. You solved?

Vogs27, 2019-09-17:
Well, I think we should change uint8_t to uint16_t at line 7157 of Marlin_main.cpp an then comment out "static_assert(Z_MAX_POS < (255- FILAMENTCHANGE_ZADD), "Z-range too high, change the T type from uint8_t to uint16_t");". I'm testing this for problems, I let you know. In the meantime, if someone more skilled than me, can confirm my thought, I'll be glad.

bogeyf14, 2019-11-06:
Did yall find a solution? I have the same problem.

Tried changing uint8 to uint16 on line 7157 but no dice. I did not try to comment out the error line. I figure it is there for a reason.

gore80, 2019-12-10:
uncomment
//#define FILAMENTCHANGE_ZADD 2
in CONFIGURATION_PRUSA_H

Demolux, 2020-02-22:
That solved it for me... while I don't know the consequences

nicolai-wachenschwan, 2022-05-30:
I think I kind of fixed it: look up the  calls of "gcode_M600_filament_change_z_shift()" by searching for "gcode_M600_filament_change_z". Then change the datatype given at this line form uint8_t to uint_16t.
At last comment the static assert, because you made the suggested change.

Prusa-Support, 2022-06-06:
Hello! Sorry for intervening so late on this thread. Sadly, I wouldn't be able to give an answer on a mod on an official capacity, but from what I see it should be just about the data type as @nicolai-wachenschwan mentioned - I wouldn't be able sadly to confirm if such mod would be safe to do or if it might incur in bugs, though - it really would have to be considered "at own risk and peril" - but I reasoned the same way as in the previous replies with the info at hand, the 365 is "outside of the 8 bit" and the error should have stemmed from that. 

Alessandro Pantaleo
Prusa Research

nicolai-wachenschwan, 2022-06-07:
Hey Alessandro, Thank you for your time! I greatly appreciate your constant support, even on old problems :)

I did the mods as I described above, so far everything works just fine. 

In the current version the function gets called only once. Within the function everything gets converted, so no problems there, and the return value gets interpreted as float, so the mod should be fairly safe. 
but I can totally understand mods deleting the warrenty :P

BladebornStudio, 2022-06-13:
In the Marlin_Main.cpp file you need to update the FilamentChange_Zadd from...

static_assert(Z_MAX_POS < (255 - FILAMENTCHANGE_ZADD)

to...

static_assert(Z_MAX_POS < (Z_MAX_POS +45 - FILAMENTCHANGE_ZADD)

Worked for me!

Prusa-Support, 2022-06-26:
Hello! Thank you for the responses! @BladebornStudio I am thinking about yours - in theory, it works, but the issue there is that you're checking the hardcoded Z_MAX_POS "against itself" and adding/subtracting something else. So I'm having doubts on what if that Z_MAX_POS might change (errors, anything that might change it in other pieces of the code by mistake, etc.). Which is the reason I believe there was the static assert "not comparing to itself" but "other values". I see it a bit riskier than other solutions to be honest. 

Alessandro Pantaleo
Prusa Research

BladebornStudio, 2022-10-11:
Makes sense, what would be a good way to go about it?

Prusa-Support, 2022-11-15:
Hello, sorry for the late response from my side! Unfortunately, I wouldn't be able to give a better solution, it would sadly be out of our knowledge, but I just wanted to point out that possible "risk" in that comparation (comparing to "itself"). Sorry for not having more knowledge on this aspect. 

Alessandro Pantaleo
Prusa Research

closed 2023-10-03 by github-actions[bot]