prusa3d/Prusa-Firmware #2068: FW 3.7.2+ planner performance regression

wavexx, 2019-08-04:
This is not strictly an issue, but mostly a head's up.

While migrating my changes to the current FW (6f3844cc00c229a190c26a4bc2324c5248a139ec) I noticed the planner cannot process instructions as fast as 3.7.1 could. In an effort to increase the gyroid printing speed I proposed an earlier patch to limit LCD refreshes to avoid starving the planner (see #1928 - which I didn't retest just yet) and modified prusaslicer to limit the complexity of the pattern (again, stuff is still pending), but in this case I'm actually seeing a reduction in the gcode throughput that we can send to the printer.

Doing a rough estimate, we could process about ~26 G1 instructions/sec in 3.7.1 in ideal conditions via USB, and we're down to about ~21 in the current master. The way I measured this is currently cumbersome: printing fixed-length segments in a pattern and increasing the speed until I could see microstutters in the resulting wall, so it's not very exact as I ultimately used a plain ruler. Still, it's enough to see large changes.

Since the main code will yield to any ISR, and the stepper ISR was not changed, this is extra load that was introduced in either the heating or lcd code. I didn't have a deep look at both yet as I was working on other stuff in the last weeks, so I'm just wondering if there's anything obvious we could do to improve the situation.

Although we're not at the point that regular prints would be affected, it definitely puts a ceiling to the speed at which some geometries can be printed. Draining the print queue during an external wall definitely shows.

michalxfanta, 2019-08-05:
Thank you for your feedback, we will look into them.

DRracer, 2019-08-05:
@wavexx this is interesting, the 3.7.2 didn't bring any major changes. Can you do a quick test with the upcoming 3.8? There is a huge patch of LCD improvements from @leptun .

leptun, 2019-08-05:
@DRracer It is possible that the current implementation of bed pwm with the isr slow down the loop. While debugging the lcd buffer I saw that an isr actually takes some time to enter and exit. This is especially evident with many consecutive isr calls where there might be little time left for the actual code before the next isr starts. Maybe lowering the frequency a bit will help. I’ll look into it later this day.

DRracer, 2019-08-05:
@leptun that's true, but FW 3.7.2 didn't have any bed PWM changes. That's why I was asking about the comparison with the upcoming 3.8.

leptun, 2019-08-05:
He said 3.7.2+ so he’s also referring to the upcoming 3.8. @wavexx pls confirm.

wavexx, 2019-08-05:
On Mon, Aug 05 2019, leptun wrote:
> He said 3.7.2+ so he’s also referring to the upcoming 3.8. @wavexx pls confirm.

I'm testing the current master, which includes the new PWM heating and I
guess all the leptun changes so far (except for the WIP PR), and
comparing this to the old 3.7.1.

I lost a bit track of the release numbers inbetween.

DRracer, 2019-08-05:
@wavexx ok, that makes sense. We may lower the bed switching frequency (higher timer prescaler). It will click in the bed, but for a quick test this may be enough.
timer02.c, line 29:
`TCCR0B = (1 << CS00);    // no clock prescaling` -> `TCCR0B = (1 << CS01); // clk/8`

wavexx, 2019-08-05:
That does bring back the previous throughput. So yes, it's the PWM heating.

I first tried to disable the bed heating after the first layer, but this barely made any difference.
Lowering the switching frequency though did work.

Spamming the LCD button and menus does have a marginal effect in boundary conditions, but so did in 3.7.1, so nothing to worry about.

leptun, 2019-08-05:
@wavexx Disabling the heating did nothing since the isr is constantly being triggered regardless of it’s necessity. I’m not sure if it’s a good idea to disable it when it’s not used as that could bring more problems than it solves and believe me it’s not fun.

wavexx, 2019-08-05:
I was hoping that it would simply exit the ISR sooner, but as you say it's really the frequency.

DRracer, 2019-08-05:
@wavexx thanks for confirmation, we'll have to play more with the timer's frequency - only do the 62kHz when necessary and then slow down. We'll have a look at it for the 3.8 final

wavexx, 2019-08-05:
On Mon, Aug 05 2019, DRracer wrote:
> @wavexx thanks for confirmation, we'll have to play more with the
> timer's frequency - only do the 62kHz when necessary and then slow
> down. We'll have a look at it for the 3.8 final

I did notice some buzzing with 3.7.1, but mostly when using a textured
sheet. With the PEI the noise is dampened.

The new code is definitely more silent, however even with the prescaler
the noise was still absent here.

Just to be clear, I'm still using the original silver PSU btw. Never had
"clicks" from the psu itself. The noise I always had/heard (if any) was
coming directly from the build-plate.

3.5 was actually pretty silent on this machine.

leptun, 2019-08-05:
Was it 3.5.2 by any chance? If yes, than you put your board under some stress because of the high frequency pwm.

DRracer, 2019-08-06:
@wavexx thanks for sharing your experience and @leptun is right - there used to be one firmware (not sure which version) doing 62kHz bed PWM. That was absolutely silent, but it wasn't too healthy for the switching MOSFET, most notably when running in an enclosure. That's why we want the fast PWM to be as short as possible.

wavexx, 2019-08-06:
On Tue, Aug 06 2019, DRracer wrote:
> @wavexx thanks for sharing your experience and @leptun is right -
> there used to be one firmware (not sure which version) doing 62kHz bed
> PWM. That was absolutely silent, but it wasn't too healthy [...]

I know the story, but I'm referring to the pre-62kHz version. I also did
use 62kHz for a while. To my ears 64kHz was relatively louder than both
the slow-pwm *and* the current version.

Still, in every iteration the bed noise was drowned by other sources
such as the part fan during printing, so I didn't really it consider
"loud" on absolute terms.

But I did look for youtube videos showing the psu-clicking, so I have an
idea on what people refer to. I guess I was lucky with this psu.

DRracer, 2019-08-06:
@wavexx yeah, me too ;) . But other machines on our farm can make a really loud sizzle from the bed.
There are also issues with PSUs singing at high frequency or making extremely loud sound slike a horn.

DRracer, 2019-08-14:
@wavexx could you please measure the performance of this? 
https://github.com/DRracer/Prusa-Firmware/tree/bed-noise-performance
... should be almost drop-in replacement for the previous version.
Please note, it's work in progress, there are still some artifacts on the output pin occasionally (so bed clicks may occur), but generally the new automaton with switching CLK prescalers is working.

wavexx, 2019-08-14:
I'll be out this week - I'll give it a good test as soon as I'm back

DRracer, 2019-08-22:
@wavexx ping ;) ... the improved bed switching code survived testing procedures correctly, now merged upstream. Could you please have a look at the performance?

wavexx, 2019-08-22:
On Thu, Aug 22 2019, DRracer wrote:
> @wavexx ping ;) ... the improved bed switching code survived testing
> procedures correctly, now merged upstream. Could you please have a
> look at the performance?

Didn't forget. I have some unexpected issues, I'm still away and it's
unlikely I can do this during this week.

DRracer, 2019-08-22:
@wavexx all right, standing by :)

DRracer, 2019-09-06:
@wavexx would you mind closing this issue?

closed 2019-09-06 by wavexx