prusa3d/Prusa-Firmware #2070: Wipe tower is way too large for the part being printed

rogerfroud, 2019-08-04:
I've just tried a trivial print in the form of an inverted button shape with soluble supports to try this out before doing something much larger.

However, the size it's automatically enforced for the wipe tower is out of all proportion to the size of the job. I see that you can change the tower width, but that just forces it to be higher to maintain the same volume. Obviously this is very wasteful. What can be done about this? It looks like there's a minimum enforced size for the job.

[image]

3d-gussner, 2019-08-05:
Maybe...if you have lot of filament changes per layer the nozzle needs to purge the defined amount. So a 5x5x5mm cube with changes in each layer will need the same purge volume as a 200x200x5mm block.

3d-gussner, 2019-08-05:
Did you try purge to infill and objects? With that small print the purge to infill isn't enough.

michalxfanta, 2019-08-05:
It actually questions for [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer), but you can lower values in Purging volumes for Wipe tower.

[image]

rogerfroud, 2019-08-05:
Ok, thanks for that. I wonder why it's set to be so large be default?

michalxfanta, 2019-08-06:
This is due to sufficient nozzle cleaning between filament changes. There is a risk of mixing colors in the model or, in the case of soluble supports, to mixed this material in the model body.

DRracer, 2020-09-21:
Closing, not a FW issue

closed 2020-09-21 by DRracer

sidekickc, 2022-06-22:
> It actually questions for [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer), but you can lower values in Purging volumes for Wipe tower.
> 
> [image]

Thanks for this. The program is super buggy still. I often have to Slice, then return to 3D Editor View, and do an Arrange, then Slice, then come back to 3D Editor View, before it shows the updated tower.  Very annoying.