prusa3d/Prusa-Firmware #2073: Mk3s filament sensor random issues filament change even when disabled.

3dmonkeynl, 2019-08-05:
Filament sensor issues random changes on printing even with the sensor disabled in the menu.
Had this a couples times I disabled it because it gave me multiple changes when using Fillamentum Vertigo Starlight.

michalxfanta, 2019-08-05:
You wrote about MK3S filament sensor, but in S version is installed IR filament sensor, which does not depend on the type of filament. And checking if the filament is present or not is mechanical. In MK3 was installed optical filament sensor, which has detected some filaments incorrectly. Which do you have?

3dmonkeynl, 2019-08-05:
I have the new IR filament sensor.

michalxfanta, 2019-08-06:
In this case, try to adjust the position of the filament sensor a little and check the sensor state switching in the menu Support > Sensor info when inserting the filament manually.

StudentSA, 2019-08-07:
Hey, Im having the same issue on my i3 MK3. Randomly in a print its been asking me to change filament??? I am on latest Firmware (3.7.2) and Slicer (2.0.0)

3dmonkeynl, 2019-08-09:
Sensor adjusted, but for small prints I don't have any issue's (1/2 hour prints) 
Will test some longer duration print to see if it pops up again.

gysnst2, 2019-09-09:
I am having major issues with my MK3s MMU2 asking for filament changes.  Right MMU2  it has 5 to choose from not The case.  the print will pause for no known issue and ask me to remove the filament and reload.  Then i hit the dial on the LCD button to continue.

closed 2023-10-03 by github-actions[bot]