prusa3d/Prusa-Firmware #2077: Z Hight wrong after cancel print

veloc1ty, 2019-08-05:
Hi,

I use an MK3S with the latest firmware and I have observed a strange behaviour. Whenever I cancel a print (pressing reset or from octoprint) the Z hight is afterwards messed up.

I have reproduced this a couple of times now and I have to go through the complete Z and first layer calibration in order to fix this.

The PINDA is perfectly placed and first layer is calibrated exactly.

Steps to reproduce:
Start any print, let it print a while
Cancel either from octoprint or by pressing reset
Home the Z axis or start a new print
Display now shows height 0.2
When starting a new print, first layer doesn't stick due to incorrect z height

Sometimes it is also too low and to close to the bed and this messed up the first layer as well and us scratching over the pei sheet.

Has anyone ever experienced the same or is there a fix available?

michalxfanta, 2019-08-06:
What firmware version are you using?

veloc1ty, 2019-08-06:
Hi,

I'm on 3.7.2, also tried re-flashing it and a factory reset.

Just had the problem again and the nozzle crashed into the bed after cancelling a print and starting another one.

veloc1ty, 2019-08-06:
Send: G28 Z0 <-- home
Send: G90

Recv: tmc2130_home_enter(axes_mask=0x01)
Recv:   0 step=62 mscnt= 995
Recv: tmc2130_goto_step 0 13 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv:   0 step=33 mscnt= 541
Recv: tmc2130_goto_step 1 33 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing

Recv: KILL: 0 <--- crashed into bed

Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error: Printer halted. kill() called!"
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"
Connection closed, closing down monitor

leptun, 2019-08-06:
What live-Z value are you using?  If it’s too close to 0, then it means PINDA is mounted too high in the holder. It also results in intermittent triggering distance since the nozzle might actually touch the bed when you have a hot PINDA (something you would expect after stopping a print)

veloc1ty, 2019-08-06:
Pinda is somewhat around 600 if I remember correctly, I have the powder coated sheet.
It is unfortunately not related to the temperature, I thought so too in the first place. But even if I cancel a print, let it cool down and then home it, it has the wrong z height.. really clueless what to do.

I am not turning off the power or doing anything stupid, pressing the "cancel" button or cancelling the print (even when it has not yet started -> slice > send to octoprint > and cancel it immediately without even moving the extruder) is resulting in the same issue almost every time.

DesC21, 2019-08-10:
I had the same that after canceling a print the Z height changed. I had no chance to reproduce it so I never posted it but apparently I am not alone anymore :).

So what happened was, I had a live-z of about -0,750. Started a print and noticed that I screwed up in the Slicer and choose the wrong filament profile. Anyway stopped the print, simply cleaned the bed and started a new print with the correct settings. I am using Ocotprint btw. Came back to the printer when the first layer was halfway done and saw the nozzle was way to close to the bed, the lines were really thick. No idea what happened. Checked the Pinda right away, is still working and hasn't moved (well not really able to accurately measure it anyway). So I calibrated the first layer again and low and behold it was +0,2... So I now have to use around -0,550... this is for the textured sheet though, so the smooth is about -0,3...

When this happened I was running firmware 3.7.2 (currently 3.8.0). So far I hasn't happened again (tried with canceling prints) but I now have this constant "offset" to my previous live-z....

leptun, 2019-08-10:
So this is octoprint related then. Can you both post the cancel script from octoprint?
@veloc1ty @DesC21

DesC21, 2019-08-10:
My cancel script is the default one:

cancel script
; disable motors
M84

;disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}
;disable fan
M106 S0

leptun, 2019-08-10:
My ending script is similar so it shouldn’t be a problem from there. I think I’ve also experienced this issue some time ago. It was right after multiple sheets was introduced. I remember that it created a first layer very close to the bed. Here is a photo I took back then. It was a PITA removing that thin layer in the middle since the edge broke away, but the center remained there. Had to print over it in order to release it with the next part. 
[image]

I thought it was just a misconfigured layer height. Maybe it’s the same issue.

wavexx, 2019-08-11:
I noticed while fixing the pause function that the world2matrix&mbl correction is not consistently added/removed in all cases. Sometimes we apply w2m _and_ mbl, sometimes we undo only mbl, and so on.

I'm not saying it's necessarily wrong, but I wouldn't be surprised if that was the case. For crashdetect to work the correction is saved and later restored without actually performing mbl again. This introduced new conditions which might not be handled correctly all the time.

These are also tricky to find, as in most cases the offsets are quite small and only add up when combined (for example, summing the mbl delta twice after a cancelled print).

wavexx, 2019-08-11:
More than crashdetect, I was speaking about powerpanic.

DesC21, 2019-08-11:
> My ending script is similar so it shouldn’t be a problem from there. I think I’ve also experienced this issue some time ago. It was right after multiple sheets was introduced. I remember that it [...]

This is exactly what it looked like for me. Could very well be the same issue ;).

@wavexx this sounds like it could be the reason. I don't know if I switched the printer off after canceling the print but on 3.7.2 I always got the power panic beep when I switched it off (via sonoff) as well as a short z move. 

So whatever calc went wrong the +0.2 offset is then stored until a complete reset right? So if I would reset it I should be back to inital live-z right?

wavexx, 2019-08-23:
A printer reset after cancelling a print should work-around this.

leptun, 2019-08-23:
I think there should be a check that sees if babystep and W2M is applied or not. This way we can try to apply or revert it as many times as we want and still have it in the right position.

leptun, 2019-08-23:
As requested by @DRracer, #2118 is where the discussion will take place.

veloc1ty, 2020-01-28:
Just a heads up, problem is still present in 3.9.0 BETA and seems to be even worse. 
Same behavior than before.

I could reproduce it a couple of times
- Level everything for a perfect first layer ™ 
- Print anything, let it run for a while 
- Cancel the print either from octoprint or cancel button on the printer (makes no difference)
- Remove object / clean heatbed
- home printer
- start another print
> first layer is messed up, nozzle way too low.

> go through first layer calibration again
GOTO 1

> Doing only calibrate Z does not make a difference
> Turning printer off and on in between does not make a difference

wavexx, 2020-01-28:
Are you using the first sheet, or some other sheet? Did you adjust live Z between prints?
I'm trying to figure out possible scenarios that could cause this

veloc1ty, 2020-01-28:
I am using the powder coated sheet, original.
It works perfectly fine once leveled in, but after cancelling a print it gets the z height wrong.

I'll try to provide more detail and maybe some measurements once I run into it again.

wavexx, 2020-01-28:
I meant in the LCD menu, you can now have multiple Z offsets assigned to each sheet.
If you go to Settings -> HW Setup -> Steel sheets you can see if you're using the first one in the set or not.

veloc1ty, 2020-01-28:
Ah Okay, sorry - No, it is not the first one, it is the texture1, the first one is smooth1 if i remember correctly - but that one is not configured.

I am not in front of the printer at the moment, but that is how it is set up.

wavexx, 2020-01-28:
On Tue, Jan 28 2020, Alex wrote:
> Ah Okay, sorry - No, it is not the first one, it is the texture1, the
> first one is smooth1 if i remember correctly - but that one is not
> configured.

As a test, try to use the first one for a while.

I noticed that my sheet sheet got reset to the first one sometimes, and
I'm not sure what did I do to make it happen. This could be a source of
offset.

This could help pinpoint the problem quite a bit.

veloc1ty, 2020-01-28:
Thanks a lot, I'll try using the first one then and update the results here.

veloc1ty, 2020-01-29:
Good Morning,

changing the sheet made a difference, I also observed that it changed back to sheet1.
I'll continue to test it and see if I can still recreate it somehow. For quite some time I had only sheet1 configured and also had the same thing a couple of times.

closed 2023-10-03 by github-actions[bot]