prusa3d/Prusa-Firmware #2079: MMU2S tries to unload filament with cold hotend after canceled print because of crash detection

langph, 2019-08-06:
I had a crash detection, opted to cancel the print and powered off the printer. When I started a new print the next day, I powered the printer back on, selected a gcode file and the exact same filament number as the day before. It was still loaded and the MMU tried to unload it with a cold hotend (which did not succeed).

michalxfanta, 2019-08-07:
Thanks for reporting this error, we will test it.

vintagepc, 2019-08-10:
I've observed this too in the 3.8.0-RC1 candidate.

DeWonderMan, 2019-11-18:
It happened to me too.

Not after a crash, but simply after printing a non MMU sliced file first.
In that case the filament is not unloaded after a succesful print.
After the print I powered off the printer for the night.

The next day I wanted to print an MMU single sliced model, and it asks which filament to use, and I choose the filament already loaded.
After that it tries to unload the filament anyway and while the hotend is still cold grinding away the filamant.

closed 2023-10-05 by github-actions[bot]