prusa3d/Prusa-Firmware #2080: Spooljoin: MMU2 failed to load next filament, but the print resumed anyway, printing with air

langph, 2019-08-06:
After one spool was depleted, the next filament failed to load. The printer resumed the print anyway.

michalxfanta, 2019-08-07:
Thanks for reporting this error, we will test it.

pjanfred, 2019-08-11:
I have the same problem. Thats the process:
1. Printing works fine
2. Filament Loading failed
3. Printer resumes printing with air
4. Printer change color and resume printing with air and so on
5. I abort the printing after 10min printing with air

The exciting thing is that the nozzle is completely cold after the break - in my case 35 degrees.

After that i try loading to nozzle manally and all is fine - after preheating.

Same FW, printer and mmu.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg