prusa3d/Prusa-Firmware #2081: Updating Firmware for PT100 & higher extruder max temp

BhagavadGitHub, 2019-08-06:
So, I haven't received my PT100 kit yet but I wanted to get started with the firmware changes. My machine is an i3 MK3s.

PT100 documentation:
https://e3d-online.dozuki.com/Wiki/E3D_PT100_Amplifier_Documentation

I started off by downloading the firmware zip here:
https://github.com/prusa3d/Prusa-Firmware

I then found the "PrusaResearch EINSY RAMBo" board on Arduino 1.8.5

Since I don't have the PT100 kit yet, I skipped the pin assignment and went to define the type 20 TEMP SENSOR (I am using Visual Studio Code to edit files).

Unfortunately I couldn't see any lines with the TEMP SENSOR. But then at the bottom, I saw listed the thermistortables.h file. This has the PT100 info listed out in line 694, which is great but how do I switch this to activate it and uncomment the current thermistor? I also am going to want to increase my max temp to 345 C and I'm not sure how this will be accomplished either.

Any help would be appreciated, if this is not the right place to post please let me know and I will move it.

Adding an additional power supply is something I would be willing to do down the line as to not overload this one, but I want to get things working first. I only plan on printing around 310 C but my understanding it that this upgrade can go to 350C, so I might as well set the limit close to there.

Thanks.

3d-gussner, 2019-08-06:
Hi, the PT100 is already in the firmware you just need to enable/uncomment the correct line in the variant you need.
I hope you also ordered a copper heat block as the default aluminium may degrade around 300°C and a better nozzle like the nozzle x or copper nozzle .
If you follow the linux subsystem for Windows or git bash for Windows you get a known build environment.
- download the latest release source code from the release tab or choose in github Branch: the latest stable Tags v3.7.2 or v3.7.3 and then press clone or download
- create a copy of the mk3s variant file in the variants folder
- edit the new copy and  uncomment the E3D_PT100_EXTRUDER line (did you get it with amp module?) it should be at line 33-37
- open linux subsystem or git bash and start the PF-build.sh script
- follow the instructions and you get your firmware build.

Hope that helps

3d-gussner, 2019-08-06:
Overload the PSU thoughts,

The default 240W PSU is already at its limits when heating up bed to 100°C at the same time heating the nozzle to 260°C it draws around  ~9.5 Amp. This means that the PSU has nearly 100% load doing both at the same time, what isn't the best for long-term... But you can tweak the sequence via gcode so the PSU doesn't have to run at this load. Example: preheat bed 1st and as it reached the temp start heating the hotend. Takes bit longer but less load on PSU.

As the Prusa already uses the most powerful heater cartridge 40W the PSU will not be overloaded...but the peak may stay longer as reaching the desired 310°C takes some time...again good start gcode can be your friend.

3d-gussner, 2019-08-06:
Goto the Prusa forum and search for PT100 you should find some info how to connect it.

BhagavadGitHub, 2019-08-06:
Awesome! Thanks for all the info. I also like this power management suggestion. I wonder if there is a way to write a startup script to heat the bed then the hot end. I will check out these resources tonight and respond if I am still having issues.

BhagavadGitHub, 2019-08-06:
Ok so I re-downloaded the firmware here: https://github.com/prusa3d/Prusa-Firmware/tree/MK3_3.7.3

My first download was a v3.72, but this time I got v3.73.

In the variants folder there are just board numbers called out (see attached).

Beyond that, neither the configuration.h file I suggested above, nor these variation files, or the thermistortables.h have any the PT100 line to uncomment.

I'm sure there is a basic mistake I am making somewhere, I am trying to be specific with the steps I took.

I'm still using Visual Studio Code, I have seen others use this with success on youtube for other prusa firmware changes, if someone can help me find another program I can give it a try. I searched git bash and wasn't sure how to get it.

The forum is suggesting editing the configuration.h file, like other sites I have seen, I just can't seem to find the line. I also still need to figure out how to change the max temp unless uncommenting the PT100 line does this already for you.

[image]

3d-gussner, 2019-08-06:
If you search for E3D_PT100 you will also find the already defined max temperatures around line 300 which is set to 410.
Again don't try it with the standard aluminium heater block it may not survive.

Also read the README.MD very carefully how to compile the firmware. Try to use PF-build.sh

Changing the start gcode in Prusa Slicer isn't hard you just need to swap a two lines so the printer waits for heat the bed 1st before you set the hotend temp. Make a new high temp profile for that otherwise you will wait longer on normal prints.

BhagavadGitHub, 2019-08-07:
Nice! Yes good suggestion, when I figure out the start script I will add it to high temp profiles only. Now when my kit comes Thursday all I have left to figure out is pin assignment. I did save a copy for the PT100.h file, so thats good to go! That part looks like that is pretty well covered in the Prusa forum. Hopefully there will be know problems and I will update this post on further progress to help any future person that struggles with this as well (and to track what all I did).

closed 2023-10-05 by github-actions[bot]