prusa3d/Prusa-Firmware #2086: [3.7.2] MMU2s unload filament fails because of extruder loading movement

workinghard, 2019-08-08:
Following observed scenario:
During filament change:
* After cooling moves, filament is being unloaded
* IR sensor triggers, extruder motor stops
* MMU2s unload command is beings send and at the same time extruder motor starts loading movement. 
* Sometimes extruder motor is faster and can caught the filament
* In this case IR sensor triggers, extruder motors stops and MMU2s reset is being initiated

Because the filament is stuck now between gears, MMU2 don't have enough force to pull the filament out and grinds it out resulting in unload issue. This is happening especially with softer PLA materials often.

Solution would be either to delay the extruder motor load movement or even do unload movement for couple seconds.

antimix, 2019-08-10:
When this happens, the print is using a MK3S MMU2 profile or MK3S single mode ?

workinghard, 2019-08-11:
I'm not using M600 (filament change based on layer height) on my MMU2s printer.
The observed scenario and the fix is for regular MMU2s color print.

culverje, 2019-10-28:
[image]

My print stops at this point every single time, I want to try:

> Solution would be either to delay the extruder motor load movement or even do unload movement for couple seconds.

My question is how do I do this, I've looked everywhere in the settings and don't see a way to do this.  Any assistance will be greatly appreciated. Thank you.

workinghard, 2019-10-28:
@culverje It's only possible if you compile the firmware yourself. You need to get the most recent firmware version and apply the pull request (patch). I'm doing it for my printer every time and will try to provide the compiled version for MK3s in a forum soon.

culverje, 2019-10-28:
> @culverje It's only possible if you compile the firmware yourself. You need to get the most recent firmware version and apply the pull request (patch). I'm doing it for my printer every time and will try to provide the compiled version for MK3s in a forum soon.

Thank you, much appreciated.

workinghard, 2019-10-28:
Here we go: https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-general-discussion-announcements-and-releases/mmu2-real-multi-material/paged/4/#post-171410

culverje, 2019-10-29:
Thank you!
:)

Mr. Culver
Fifth Grade Teacher
Silver Lake Elementary
5858 Culver Rd.
231-933-3651
culverje.weebly.com

lilygomes, 2021-01-09:
[image]
This happens to me with every print: failed unload of filament 4 (1: Prusament PETG, 4: PolyLite PLA, 5: PolyLite PLA) and a whole of of PLA shavings with no actual movement. Can't remove the filament with pliers when I remove the festo fitting. Always fails at the same point in my print.

antimix, 2023-09-29:
Has been this fixed ?

culverje, 2023-09-29:
I just disassembled it, maybe I'll try the MMU3. Thanks

closed 2023-12-11 by github-actions[bot]