prusa3d/Prusa-Firmware #2091: Feature Request: Allow uncalibrated sheet to be selected by manually specifying live-z

vintagepc, 2019-08-10:
Feature request for the sheets menu:
Currently, an uncalibrated sheet cannot be selected and you must run the first layer cal on it to use it. 

Please allow selection/use of an uncalibrated sheet via manually entering the offset. Prior to this feature I and many others kept track of our sheets with labels and relative offsets; e.g. I know my smooth sheet is 0.350 thicker than my PC sheet, and they both have a label with their Z values. Being able to calibrate a single sheet and then just dial-in any others based on that value without re-doing the calibration would be a bonus. 

Similarly, sometimes a factory reset is necessary, and if the printer has not been altered, the previous Live-Z values were still valid and it would be nice to be able to just re-enter them and resume printing without needing to possibly swap to PLA, run the calibration, and then change back again to whatever you were using before.

maximlevitsky, 2019-08-13:
I would like that as well. Meanwhile there is a workaround. You can select live Z calibration from the sheet menu, and then cancel it (just say that PLA is not loaded or so), and then the sheet will be selected, so you can change the live Z in the settings menu.

michalxfanta, 2019-08-29:
We understand the reasons why to have direct z-offset entry, but for safety reasons, it will still be necessary to perform First Layer Calibration. If you know z-offset for each sheet, you could set this value at the beginning of the first layer calibration and wait to the end. It should not be a problem, because you will not calibrate the first layer often.

closed 2019-08-29 by michalxfanta

vintagepc, 2019-08-29:
I have to disagree. 
I'm not seeing how this is any more or less of a safety issue than having the wrong sheet selected (#2093), allowing the user to print with a live Z of zero, or any number of other things they can do with the printer. There is no "safety" to be gained by forcing a live Z calibration over a user entry if the default is zero. The user can mess up a live Z cal just as easily as they could enter a wrong number; It just feels like unnecessary user hand-holding that gets in the way, especially for those that may not use PLA often or at all (It's somewhat less of a pain point if other options become available for firs layer cal).

GurliGebis, 2019-08-29:
I agree, having to use PLA just to be able to calibrate the sheets.
Most people with multiple sheets already have them calibrated, and the offset written down. Why not just let people enter them manually.

kevinkahn, 2019-09-07:
There is no safety issue here. Some of us don't normally use PLA. This is pointless user control that makes life hard for the advanced user for no reason.

Area5142, 2019-09-07:
I agree, 95% of my prints is made in PETG and I use my own calibration G-Code for PETG.
I would like to avoid having PLA into the nozzle and the risk of carbon build up (clogs) when temperature again is raised for PETG.

draeath, 2019-09-08:
I also do not agree with the safety concern. It is no less safe than immediately going into settings and raising the adjust after the forced initial run. It does nothing to stop us putting wrong values in there, so why should it do so here?

hoppke, 2019-09-11:
@michalxfanta changing nozzle, wiping fail stats, moving the printer (and/or running xyz cal), Y axis or frame maintenance, accidentally bumping into the pinda (e.g. whilst taking off the flex sheet), calibrating extrusion for a new filament/filament type... any of these warrants running first layer cal or wipes it from printers memory.
The thing Prusa missed out on here is that users do what @vintagepc describes - we've a list of delta offsets between our sheets, and maybe an additional list of deltas for filaments. Sure, I've 3 sheets, but if I change the nozzle I'll only need to get first layer right on one of them, the remaining two fall into place via the deltas I've written down somewhere.

What would make my UX better right now would be custom gcode for writing sheet data. Sth like "Mxxx sheetIndex label offset"...

GurliGebis, 2019-09-13:
@vintagepc please make a pull request for it - it looks great :) (Haven't tested it, since I haven't got the time atm.)