prusa3d/Prusa-Firmware #2093: proposal: position of the sheet selection menu

DesC21, 2019-08-10:
I think it would be nice if you could see which sheet is selected from the main menu before starting a print. When I start a print from the sd card, I have to scroll down to check if the correct sheet is selected. If the menu entry for the sheet selection was higher to the top, for example directly before or after "PRINT FROM SD" (I'd prefer after), the user could immediately see if the print can be started.

vintagepc, 2019-08-10:
I would advocate for something more visible too; I also nearly did a number on my smooth sheet when I'd accidentally had the printer set to the PC profile. A prompt prior to starting the print could be problematic in the octoprint/remote workflow but perhaps finding a way to display the selected sheet on screen during warmup and pre-heating with the X/Y/Z (X0/Y210/Z50?) in such a position that it's easy to quickly change sheets would be viable.

jbinfl, 2019-08-18:
Ohh, I like this suggestion and vintage PCs idea.  that would be very helpful if possible to implement!

TehH4rRy, 2019-08-28:
This has my vote! I've been lucky to only scuff my PEI by not checking which sheet I have on

vintagepc, 2019-08-29:
I decided to implement it on the info screen by replacing the percent done when either the printer is heating or there is no percentage (idle). Both regions consume exactly 7 characters, so it works very nicely.

DesC21, 2019-09-07:
perfect, thanks Prusa Team :-)!

closed 2019-09-07 by DesC21