prusa3d/Prusa-Firmware #2097: 2..0 Error Exporting Config Bundle

AbeFM, 2019-08-10:
I exported a config bundle in my release version of 2.0, when I tried to load the bundle into 2.1a, I got the following error:

[image]

Looking at the export, checking the specific line, I found that to be the last in the file:

[image]

From this I infer my export didn't work right. Curiously, PS2.0 exits during/after the config bundle export, perhaps it is crashing, I do not know its intended behavior. It indicates it is exporting, then blinks out of existence after a short pause.

vintagepc, 2019-08-11:
You should report this to https://github.com/prusa3d/Prusaslicer/issues as it's a slicer issue, not a firmware issue.

AbeFM, 2019-08-12:
That's.... right.

Thanks. :-)

closed 2019-08-12 by AbeFM