prusa3d/Prusa-Firmware #2098: MMU-Different Filament Temperatures

FloatingCameras, 2019-08-10:
# Problem:
Currently the Hotend temperature changes to the next filament's temperature _at the start_ of the retration.  **This should not occur until the filament is clear of the extruder**.
# Reason:
The type of tip created during retraction is determined by temperature.  I've found when the temperature is too hot the filament tip is a long string or worse, a bulbous string.  These cause load and unload failures.
# Example:
- Filament #1: 225
- Filament #2: 230
- Filament #3: 235

In the configuration above Filament #1 will be retracted at 230 and 235 depending on which filament is used next.

# Observation:
At 225 the tip formed on Filament #1 is a short string and does not cause load failures.  At 230 or higher it forms a long string and loading failures occur often. 

I have found when I am experiencing lots of loading errors from a filament if I trim the string off the tip manually after it is retracted and before it is used again there are no load failures. As soon as I just let it run there is a load failure within a few filament changes.  The tip is really important!
# Summary:
**The filament should be retracted at a known temperature**. 

# Solution:
- Easy fix:  Wait until the filament is clear of the extruder to change temperature so the filament temperature is known and always the same during retraction.
- Complex fix: Add a retraction temperature parameter to the filament settings.  This would allow printing at a higher temperature and always use the best retraction temperature for that filament.

vintagepc, 2019-08-11:
This is an issue for Prusa-slicer as that's all done in gcode, not firmware.

FloatingCameras, 2019-08-11:
Thanks.  I'll  look into it further and submit there.

closed 2019-08-11 by FloatingCameras