prusa3d/Prusa-Firmware #2099: Factory reset not clearing all data (3.8.0-RC)

vintagepc, 2019-08-11:
I experienced this recently after changing firmwares; I was running a modified firmware with altered microstep values for X/Y. I flashed the 3.8.0-RC and then did a factory reset, clearing *all* data after a reboot. It did something as I was presented with the first-start wizard when it rebooted after finishing.

Much to my surprise though, M350 reported that the microstep values for X and Y were still fixed at 8 and not the firmware default of 16. I had to apply the patch for M350 (see #2024) in order to correct this. Other things (Pinda temperature table, PID, steps/mm, first layer calibrations) were reset properly.

leptun, 2023-05-03:
Is this still an issue?

vintagepc, 2023-05-03:
Should be easy to verify and reproduce with the sim if it is. :)

Prusa-Support, 2023-06-29:
Hey guys, were you able to confirm if this is still a thing?
@leptun @vintagepc 

Michele Moramarco
Prusa Research

closed 2023-10-05 by github-actions[bot]