prusa3d/Prusa-Firmware #2101: Knob clicks unresponsive after inadvertant second long push while in Z-Axis move procedure

RodKimmell, 2019-08-12:
Example of printer hang...
1.  From the Main Menu, long press knob to gain access to Z-Axis move via the shortcut.
      [Display now shows " Z:        +xx.x"]
2. While in the Z-Axis move, long press button again.
       [Display still shows " Z:        +xx.x"]  
3. Rotating knob moves Z axis but clicking the knob does not exit procedure. 

Knob is forever non-responsive to clicks even after the procedure times out and returns to the Main menu.

Only know work-around is power cycling the printer.

Refer to #1924, may be related.

leptun, 2019-08-12:
A fix to this is already in a PR by @wavexx. #1759 

@DRracer maybe this PR could be merged quicker. Lets say before 3.8 final. What do you say? It’s a minor fix that shouldn’t require much testing.

RodKimmell, 2019-09-21:
This bug exists in FW 3.8.0.   

@michalxfanta please add a label for 3.8.0.

leptun, 2020-01-26:
The issue is no longer present in the latest 3.9.0-beta. You can close the issue now

closed 2023-10-05 by github-actions[bot]