prusa3d/Prusa-Firmware #2102: 3.8.0 RC1 - New Bed Heating Sound Behavior

vertigo235, 2019-08-12:
OK yes, another "clicking" sound with the bed heating issue.

In the past, clicking could be heard while the bed was heating up, this was not a big deal to me, but with 3.8.0 RC1, my bed / PSU clicks constantly once it is *maintaining* temperature.  This is much worse and more undesirable IMO.

Is anyone else experiencing the same?

vertigo235, 2019-08-12:
https://youtu.be/Fx1k3Q91QQo

MakeSomeGood, 2019-08-13:
Yup, it's clicking is worse with the newest firmware (3.8.0 RC1). 

We've been discussing it here: [https://github.com/prusa3d/Prusa-Firmware/issues/1862](https://github.com/prusa3d/Prusa-Firmware/issues/1862), but starting a new issue might be better since that other issue was both the PSU and the heatbed clicking.

DRracer, 2019-08-14:
@vertigo235 is it really the bed or is it the PSU? From the latest reports even in the 110V/60Hz countries the bed was silenced, the sound was coming from the PSU (clicking).
What power source your connected to? 110V/60Hz? What PSU do you have (silver/black)?

vertigo235, 2019-08-14:
It is the PSU and I'm on 110V silver PSU yes.

DRracer, 2019-08-14:
@vertigo235 thanks for your confirmation ... I'm glad to hear, that the bed is silent. We're working on the T-Led PSU & 110V/60Hz issue.

DRracer, 2019-08-22:
@vertigo235 we may have found the cause, the solution will be in 3.8 RC2, please stay tuned.

vertigo235, 2019-08-22:
Standing by! :D

vertigo235, 2019-08-22:
I will build this latest HEAD and see if it resolves it.

DRracer, 2019-08-22:
@vertigo235 do it now (use branch MK3), we just merged all PRs upstream for the RC2, testing now...

vertigo235, 2019-08-22:
So the silver PSU needs it's own variant now? uncomment the PSU_Delta define it appears?

vertigo235, 2019-08-22:
hmm this looks unrelated, something about the z axis, interesting

vertigo235, 2019-08-22:
Much improved, there was a little hint of a few clicks as it leveled off, but mostly silent now with an occasional little chirp. I suspect that wont be noticeable once it is printing.

DRracer, 2019-08-23:
@vertigo235  thank you for confirmation, I'm very happy the new version (almost) solves the problem.

cimoalpacino, 2019-09-11:
Hi, I would just like to ask if this piece of code in function `timer0_init()` in **timer02.c** is correct:
`TCCR2B = (4 << CS20); //WGM_2=0, CS_0-2=011`?

Does the code above mean that we set CS22 to 1 or was it a mistake and should be 1<<CS20 instead?
Because:
`CS20 = 1`: no prescaling
`CS22 = 1`: clk/128 (from prescaler)

The comment `CS_0-2=011` at the end of the code is also a bit confusing.
Does that mean 
```
CS20 = 0
CS21 = 1
CS22 = 1
```
or
```
CS20 = 1
CS21 = 1
CS22 = 0
```
With first bit configuration the datasheet states that we should get clk/256 prescaler and with second configuration we get a clk/32 prescaler.
Could someone take a look at that code if everything is OK and give me feedback so that I could understand better?
Thank you.
regards.

DRracer, 2019-09-12:
@cimoalpacino you are right, the comments are misleading. Timer2 should now work the same way timer0 was before FW 3.8, i.e. provide the millisecond clock and related functions.

cimoalpacino, 2019-09-21:
@DRracer, I'm still confuced about that piece of code in the TCCR2B register - I think there might be something wrong here.

I made a simple code on [www.cpp.sh](url). 
Setting `(1 << CS20)` or `(4 << CS20)` gives a different prescaler.
```cpp
#include <iostream>
#include <string>
#define CS20 0
#define CS21 1
#define CS22 2
#define WGM22 3
#define FOC2B 6
#define FOC2A 7`>
//assumes little endian
void printBits(size_t const size, void const * const ptr)
{
    unsigned char *b = (unsigned char*) ptr;
    unsigned char byte;
    int i, j;
    for (i=size-1;i>=0;i--)
    {
        for (j=7;j>=0;j--)
        {
            byte = (b[i] >> j) & 1;
            printf("%u", byte);
        }
    }
    puts("");
}

int TCCR2B = 0;

int main()
{
    // Set (1<<CS20)
    TCCR2B = (1<<CS20);
    printBits(sizeof(TCCR2B), &TCCR2B);   // --> Output: 0b00000001

    // Set (4<<CS20)
    TCCR2B = 0;
    TCCR2B = (4<<CS20);
    printBits(sizeof(TCCR2B), &TCCR2B);   // --> Output: 0b00000100
}
```

I'm wondering if the intention was to have no prescaling (in case of `(1<<CS20)`) or a `clk/64` prescaler (in case of `(4<<CS20)`, which is identical if we would set `(1<<CS22)`)?
I'm still quite doubtful that everything is correct, so I'd like to hear your opinion.
Thank you.
Regards.

closed 2019-10-07 by michalxfanta