prusa3d/Prusa-Firmware #2108: after runout, MMU2s fails to heat nozzle before attempting to load new filament

aarchiba, 2019-08-14:
If the filament runs out (spurious in my case but the sequence is the same), the MMU2S retracts, moves the selector aside, ejects the current filament, and prompts the user to load a new filament. When the user does this, they get the usual "is filament loaded?" prompt on the LCD, but since the nozzle has been let cool down, the extruder won't turn and the filament is never successfully loaded, let alone running clear. Even if one hits "yes" and the print resumes, the loading process is never completed. Correct behaviour would be to return the nozzle to temperature before attempting to load a new filament.

michalxfanta, 2019-08-23:
Which firmware version do you use?

aarchiba, 2019-08-27:
MK3S 3.7.2-2363
MMU2 1.0.5-297

closed 2023-10-05 by github-actions[bot]