prusa3d/Prusa-Firmware #2110: M125 Support?

icanhazpython, 2019-08-15:
I've been using the m600 filament change command to pause the printer so i can insert nuts into the print. I don't actually change the filament during this process, so not needing to go through the whole unload/load/purge sequence in this case would be beneficial to me. I see that the marlin firmware supports an M125 command which pauses and parks the head in the desired location. Is there any way to get this feature supported for prusa? I searched the firmware here and see no mention of this option. Here is where it is mentioned in marlin: https://github.com/MarlinFirmware/Marlin/blob/85c82ab28e13cb25043ae166062785bff39434b9/Marlin/Configuration_adv.h#L975

If there is a gcode sequence I can use instead which others have good luck with, I would be happy to go that route as well. I saw this suggested somewhere but have not tested yet:
```
G91                  ; Put in relative mode
G1 Z10               ; Lower bed by 10mm
G90                  ; Put back in absolute mode
G1 X0 Y0             ; Zero (home) the X & Y
M300 S300 P1000      ; play beep sound to signal attention
M0 Click To Restart  ; Pause and wait for the user
G91                  ; Put in relative mode
G1 Z-10              ; Raise the bed back up 10mm
G90                  ; Put back in absolute mode
```

icanhazpython, 2019-09-01:
I wanted to add that I tested the above code and it works well for my needs. My only gripe is that it only beeps once. Having it beep repetitively as M600 does would be great.

cimoalpacino, 2019-09-01:
Use M601 to pause and M602 to resume.
Implement it in layer change gcode script.

wavexx, 2019-10-20:
We could alias M125 to M601. The pause machinery already supports the required arguments, although they're not parsed.

closed 2023-10-05 by github-actions[bot]