prusa3d/Prusa-Firmware #2112: [3.7.2] MMU [1.0.5] more filament change tries before error

R0m3oF0x, 2019-08-16:
During the filament change, after two unsuccessful attempts, further malfunctions occur which can only be corrected manually. The only thing the operator has to do is press the knob.

There is an urgent need for a way to increase the number of filament change attempts from 2 to more, which can be set via the control panel.

leptun, 2019-08-16:
The number of tries is changed in the file linked before. Search for tries, change accordingly for load and unload, compile the firmware and upload to the MMU2(S) unit.

https://github.com/prusa3d/MM-control-01/blob/master/MM-control-01/motion.cpp

gudnimg, 2023-07-27:
FW 1.0.5 is already out of date. Last FW is 1.0.6 so I feel we should close this issue as it's "stale"

closed 2023-07-27 by gudnimg